ouster-lidar/ouster-sdk #451: Link down error (ptp4l issue)

bobzwik, 2022-11-09:
Hello, sorry in advance for the long post.

We've had an issue com up twice now, once in April, once yesterday. These issues happened on different drone platforms. 
The first one using:
- AAEON PICO-WHU4 (i5-8365UE)
- Ubuntu 18.04 w/ ROS Melodic
- Ouster OS1 (RevD perhaps)

The second platform using:
- AEEON PICO-TGU4 (i5-1145G7E)
- Ubuntu 20.04 w/ ROS Noetic
- Ouster OS0 Rev6

## The error:
A few minutes after boot and initialization, this error occurs: 
```
Nov 07 14:13:11 nvidia-PICO-TGU4 ptp4l[1359]: [916.803] port 1: link down
Nov 07 14:13:11 nvidia-PICO-TGU4 ptp4l[1359]: [916.803] port 1: MASTER to FAULTY on FAULT_DETECTED (FT_UNSPECIFIED)
Nov 07 14:13:11 nvidia-PICO-TGU4 ptp4l[1359]: [916.803] selected local clock 000732.fffe.94f54b as best master
Nov 07 14:13:11 nvidia-PICO-TGU4 ptp4l[1359]: [916.803] assuming the grand master role
```
The Ouster stops publishing its data, which causes our odometry to fail. The Ouster restarted to publish data 6 seconds later, but by then our drone had moved to much and the odometry was panicking.

Unfortunately, this only happened once on each platform, after over 20-30 hours of operational time. Without knowing the cause, it will be hard to reproduce this error, and it is unlikely for this error to reoccur naturally anytime soon. We were wondering if anyone recognizes this error and would have an idea of what could be the issue.

## Note 1:
We haven't done the [ptp4l and phc2sys configuration](https://static.ouster.dev/sensor-docs/image_route1/image_route2/appendix/ptp-quickstart.html#) as provided by Ouster. Either the instruction came out after we established our procedure, or we totally missed it. We start `ptp4l` in our code, instead of as a service on boot like in your procedure. We also haven't touched `phc2sys`. **Could our issue be linked to this?** We're trying to get more data on this glitch before following your configuration procedure.

## Note 2:
Here is our network capabilities:
```
nvidia@nvidia-PICO-TGU4:~$ ethtool -T eno1
Time stamping parameters for eno1:
Capabilities:
    hardware-transmit     (SOF_TIMESTAMPING_TX_HARDWARE)
    software-transmit     (SOF_TIMESTAMPING_TX_SOFTWARE)
    hardware-receive      (SOF_TIMESTAMPING_RX_HARDWARE)
    software-receive      (SOF_TIMESTAMPING_RX_SOFTWARE)
    software-system-clock (SOF_TIMESTAMPING_SOFTWARE)
    hardware-raw-clock    (SOF_TIMESTAMPING_RAW_HARDWARE)
PTP Hardware Clock: 1
Hardware Transmit Timestamp Modes:
    off                   (HWTSTAMP_TX_OFF)
    on                    (HWTSTAMP_TX_ON)
Hardware Receive Filter Modes:
    none                  (HWTSTAMP_FILTER_NONE)
    all                   (HWTSTAMP_FILTER_ALL)
    ptpv1-l4-sync         (HWTSTAMP_FILTER_PTP_V1_L4_SYNC)
    ptpv1-l4-delay-req    (HWTSTAMP_FILTER_PTP_V1_L4_DELAY_REQ)
    ptpv2-l4-sync         (HWTSTAMP_FILTER_PTP_V2_L4_SYNC)
    ptpv2-l4-delay-req    (HWTSTAMP_FILTER_PTP_V2_L4_DELAY_REQ)
    ptpv2-l2-sync         (HWTSTAMP_FILTER_PTP_V2_L2_SYNC)
    ptpv2-l2-delay-req    (HWTSTAMP_FILTER_PTP_V2_L2_DELAY_REQ)
    ptpv2-event           (HWTSTAMP_FILTER_PTP_V2_EVENT)
    ptpv2-sync            (HWTSTAMP_FILTER_PTP_V2_SYNC)
    ptpv2-delay-req       (HWTSTAMP_FILTER_PTP_V2_DELAY_REQ)

nvidia@nvidia-PICO-TGU4:~$ ethtool -T enp3s0
Time stamping parameters for enp3s0:
Capabilities:
    hardware-transmit     (SOF_TIMESTAMPING_TX_HARDWARE)
    software-transmit     (SOF_TIMESTAMPING_TX_SOFTWARE)
    hardware-receive      (SOF_TIMESTAMPING_RX_HARDWARE)
    software-receive      (SOF_TIMESTAMPING_RX_SOFTWARE)
    software-system-clock (SOF_TIMESTAMPING_SOFTWARE)
    hardware-raw-clock    (SOF_TIMESTAMPING_RAW_HARDWARE)
PTP Hardware Clock: 0
Hardware Transmit Timestamp Modes:
    off                   (HWTSTAMP_TX_OFF)
    on                    (HWTSTAMP_TX_ON)
Hardware Receive Filter Modes:
    none                  (HWTSTAMP_FILTER_NONE)
    all                   (HWTSTAMP_FILTER_ALL)
```

The Ouster is on our `enp3s0` port, and its capabilities look identical to the one shown in the Ouster procedure. The `eno1` port is used to communicate with our ground station's laptop. We already use `chrony` to set our PICO computer (on the platform) as master, and the laptop synchronizes its clock with the PICO's. Does this sound adequate?

## Note 3:
We are technically able to reproduce the same `link down` error (all 4 error lines) by unplugging the ethernet cable between the Ouster and the PICO computer. We doubt that a faulty ethernet connection is the cause (everything still worked fine even with the connector half-out of the ethernet port), but could EM noise cause a glitch in the ethernet transmission? We're using a flat unshielded ethernet cable, which passes near our 40VDC power distribution board (I know it sounds bad, but the cable is only 15cm long, and this glitch happened only twice in the 1.5 years we've had our drones). Otherwise, the ethernet transmission passes through a slipring between the Ouster and PICO, which is technically another transmission mechanical point-of-failure. **In your opinion, can the `link down` error occur from a programming glitch, or are we on the right track and this error can only be caused by mechanical failure in the transmission or by EM interference?** Which would be strange as our drone has flown hours without issue.

I don't know how helpful this is, but the following is the Ouster error log which occurs before and after we unplug and plug the ethernet cable between the PICO and Ouster. Unfortunately, we weren't logging these errors when the glitch naturally occurred. 

Before unplugging:
```
{"log": 
[
{"active": true, "category": "ETHERNET_LINK_BAD", "cursor": 0, "id": "0x01000011", "level": "WARNING", "msg": "Ethernet link bad, please check network switch and harnessing can support 1 Gbps Ethernet.", "msg_verbose": "Link transitioned to 0/Unknown", "realtime": "19036610892"},
	
{"active": false, "category": "ETHERNET_LINK_BAD", "cursor": 1, "id": "0x01000011", "level": "WARNING", "msg": "Ethernet link bad, please check network switch and harnessing can support 1 Gbps Ethernet.", "msg_verbose": "Link transitioned to 1000/Full", "realtime": "21126388460"}
], 

"next_cursor": 2, "active": []}
```
There is sometimes these 2 warnings, and sometimes 4 (which are duplicates of these 2 warnings)

After unplugging and replugging:
```
{"log": 
[
{"active": true, "category": "ETHERNET_LINK_BAD", "cursor": 0, "id": "0x01000011", "level": "WARNING", "msg": "Ethernet link bad, please check network switch and harnessing can support 1 Gbps Ethernet.", "msg_verbose": "Link transitioned to 0/Unknown", "realtime": "19036610892"}, 

{"active": false, "category": "ETHERNET_LINK_BAD", "cursor": 1, "id": "0x01000011", "level": "WARNING", "msg": "Ethernet link bad, please check network switch and harnessing can support 1 Gbps Ethernet.", "msg_verbose": "Link transitioned to 1000/Full", "realtime": "21126388460"}, 

{"active": true, "category": "ETHERNET_LINK_BAD", "cursor": 2, "id": "0x01000011", "level": "WARNING", "msg": "Ethernet link bad, please check network switch and harnessing can support 1 Gbps Ethernet.", "msg_verbose": "Link transitioned to 0/Unknown", "realtime": "1667945774689636608"}, 

{"active": true, "category": "UDP_TRANSMISSION", "cursor": 3, "id": "0x01000016", "level": "WARNING", "msg": "Could not send lidar data UDP packet to host; check that network is up.", "msg_verbose": "Failed to send lidar UDP data to destination host 192.168.10.1:40297", "realtime": "1667945774736348160"}, 

{"active": true, "category": "UDP_TRANSMISSION", "cursor": 4, "id": "0x01000019", "level": "WARNING", "msg": "Could not send IMU UDP packet to host; check that network is up.", "msg_verbose": "Failed to send imu UDP data to destination host 192.168.10.1:47483", "realtime": "1667945774736993280"}, 

{"active": false, "category": "ETHERNET_LINK_BAD", "cursor": 5, "id": "0x01000011", "level": "WARNING", "msg": "Ethernet link bad, please check network switch and harnessing can support 1 Gbps Ethernet.", "msg_verbose": "Link transitioned to 1000/Full", "realtime": "1667945776781163264"}
], 

"next_cursor": 6, "active": 
[
{"active": true, "category": "UDP_TRANSMISSION", "cursor": 3, "id": "0x01000016", "level": "WARNING", "msg": "Could not send lidar data UDP packet to host; check that network is up.", "msg_verbose": "Failed to send lidar UDP data to destination host 192.168.10.1:40297", "realtime": "1667945774736348160"}, 

{"active": true, "category": "UDP_TRANSMISSION", "cursor": 4, "id": "0x01000019", "level": "WARNING", "msg": "Could not send IMU UDP packet to host; check that network is up.", "msg_verbose": "Failed to send imu UDP data to destination host 192.168.10.1:47483", "realtime": "1667945774736993280"}
]

}
```

**Is the `Ethernet_link_bad` warning something we should be trying to fix?**

bobzwik, 2022-11-09:
> The `Ethernet_link_bad` error mainly indicate an issue with the cable or the cable connectors. I recommend reaching out to [ouster customer support](https://ouster.atlassian.net/servicedesk/customer/portal/8) for a better advice about this issue.

Thanks, will do. We will also switch up the cable we use and take the slipring out of the equations, to see if those warnings still occur.

Have you encountered the `link down` error before? Do you know if it is a software related issue, or a hardware/connection issue?

Samahu, 2022-11-09:
@bobzwik I have experienced this error `Ethernet_link_bad` myself on one of the sensors that I am using. The way I resolve this error typically by ensuring the cable is plugged properly and then the error goes away; which suggests it is mainly a hardware problem. This being the case I think our technical support would probably know more about the issue and what are the best ways to address it.

Samahu, 2022-11-10:
Closing the issue due to most likely being a hardware problem.

closed 2022-11-10 by Samahu