ouster-lidar/ouster-sdk #452: PCap to Visualization

sanehk, 2022-11-10:
Hi~

I'm trying to Visualization from PCAP, Json data.
(Env : Windows, Visualstudio 2022, Qt 5.12.2, c++)

Now I have some problem with visualization.
MyTest visualization displays wierd view. 

1. View on OusterStudio (same Pcap file)
[image]

2. View on MyTest App (same Pcap file)
[image]

3. MyTest code

    auto handle = sensor_utils::replay_initialize(strPCap);
    auto info = sensor::metadata_from_json(strJson);

    size_t w = info.format.columns_per_frame;
    size_t h = info.format.pixels_per_column;

    auto scan = LidarScan(w, h, info.format.udp_profile_lidar);

    get_complete_scan(handle, scan, info);

    XYZLut lut = make_xyz_lut(info);
    auto range = scan.field(sensor::RANGE);
    auto cloud = cartesian(range, lut);

    viz::PointViz viz("Viz test");
    viz::add_default_controls(viz);

    size_t cloud_size = cloud.rows();

    std::vector <float> points;
    std::vector <float> colors;

    auto clouds = std::make_shared <ouster::viz::Cloud>(cloud_size);
    viz.add(clouds);

    // auto reflect = scan.field(sensor::REFLECTIVITY);

    for (int i = 0; i < cloud.rows(); i++)
    {
        auto xyz = cloud.row(i);
        if (!xyz.isApproxToConstant(0, 0))
        {
            points.push_back(xyz(0));
            points.push_back(xyz(1));
            points.push_back(xyz(2));

        }
    }

    clouds->set_xyz(points.data());
    // clouds->set_key(colors.data()); reflect??

    viz.update();
    viz.run();

If I export cartesian coordinates from MyTest App to 'CSV' file, and open it another PointsCloud viewer (CloudCompare)
It displays well.

Am I missing some process to displaying by 'ouster::viz' ?
And I wish to let me get some c++ example to visualization by using 'SIGNAL' or 'REFLECTIVITY' with XYZ coordinates.

Thank you.

kairenw, 2022-11-10:
Hi @sanehk,

Do you simply need to visualize with `SIGNAL` or `REFLECTIVITY` using a utility, or do you need to write your own C++ visualization for it? If you simply need a utility, we recommend the [visualizer available with the Python SDK](https://static.ouster.dev/sdk-docs/sample-data.html#visualize-it).

In any case, the reason you're getting this is that [`set_xyz`](https://github.com/ouster-lidar/ouster_example/blob/master/ouster_viz/include/ouster/point_viz.h#L522) expects the data in different order:

```    
     * Set the XYZ values.
     *
     * @param[in] xyz pointer to array of exactly 3n where n is number of
     * points, so that the xyz position of the ith point is i, i + n, i + 2n
```
Whereas you're putting the xyz position of the ith point in `i*3, i*3+1, i*3+2`, if that makes sense

sanehk, 2022-11-10:
@kairenw 
I want to visualize `SINAL` or `REFLECTIVITY` using `ouster_viz` library in my C++ project.

I'll change `set_xyz` order.

Thank you~

closed 2022-11-11 by sanehk