ouster-lidar/ouster-sdk #458: spdlog dependency not documented.

themightyoarfish, 2022-11-23:
When building current master on macos arm

```
CMake Error at ouster_client/CMakeLists.txt:5 (find_package):
  By not providing "Findspdlog.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "spdlog", but
  CMake did not find one.

  Could not find a package configuration file provided by "spdlog" with any
  of the following names:

    spdlogConfig.cmake
    spdlog-config.cmake

  Add the installation prefix of "spdlog" to CMAKE_PREFIX_PATH or set
  "spdlog_DIR" to a directory containing one of the above files.  If "spdlog"
  provides a separate development package or SDK, be sure it has been
  installed.
```

There does not seem to be a hint that this needs to be installed anywhere.

Samahu, 2022-11-23:
@themightyoarfish Unfortunately we haven't updated the documentation. Will keep this ticket open till then.

kairenw, 2023-02-17:
The documentation has been updated. Closing issue.

closed 2023-02-17 by kairenw