ouster-lidar/ouster-sdk #461: Consider not defaulting to EIGEN_MAX_ALIGN_BYTES=32

themightyoarfish, 2022-11-23:
https://github.com/ouster-lidar/ouster_example/blob/279c9fb36c7e217d76bdde56bc82f833fe14783b/CMakeLists.txt#L23

I believe this is rather dangerous. I am currently debugging a case where I have compiled PCL from source on macOS ARM (m1 cpu with `-mtune=native`). This does not set any specific eigen alignment requirements, and I suspect that Eigen's default alignment for this platform is 16 – at least the pointers are 16 byte aligned, but not always 32.

Now when my client project, which uses both PCL and ouster-sdk is built, it gets this 32 byte alignment requirement mixed in by ouster. But the PCL library I link with does not know this and may create unaligned data. This leads to a world of pain (e.g. crashes when the unaligned data gets deleted).

So it appears to me that defaulting this option to on will break on any system where the default alignment is 16. I have confirmed that the problem does not arise if I disable this option in the ouster build, so it does not leak into my project.

Samahu, 2022-11-28:
Hi @themightyoarfish , I see your point maybe we better off granting library users the ability to set the alignment requirement of eigen to their liking rather than being fixated to 32 bytes.

themightyoarfish, 2022-12-09:
Particularly as this [PR here](https://github.com/ouster-lidar/ouster_example/pull/420/files#diff-922dacb8ece23fe3a50d783a813befd2ed9328a238dd8b7f911f7c75b8956db4L73) removed that setting for `ouster_ros`  – I'm assuming for this reason.