ouster-lidar/ouster-sdk #462: Replacement for "int replay(playback_handle& handle, double rate)"

bk-wilson, 2022-11-24:
Hello all,

I recently updated from 20210608 to 20220927 and noticed the pcap replay function I was using has been removed

I have been searching through the code and and cannot find anything obvious that replaces this feature.

How does one use "ouster_pcap" now?

Regards,

brian

Samahu, 2022-11-24:
Did you take a look at the `lidar_scan_example.cpp`:
https://github.com/ouster-lidar/ouster_example/blob/master/examples/lidar_scan_example.cpp

The python code also has an example that shows how to replay pcap files:
https://github.com/ouster-lidar/ouster_example/blob/279c9fb36c7e217d76bdde56bc82f833fe14783b/python/src/ouster/pcap/pcap.py#L271

bk-wilson, 2022-11-24:
Thanks @Samahu,

I did look into `lidar_scan_example.cpp` but it's not exactly what I needed.

We have two applications, the main application that implements the client and a secondary application that replays pcap files using `ouster_pcap`.

My solution for now is to update the replay application with our own calls to `sendto`.

Would it be fair to say the intention now by Ouster is to embed any replay features needed directly in the main application?

Regards,

Brian

kairenw, 2022-11-24:
Okay I see, you were using `replay`. Yes, what happened is that we introduced the Python SDK and moved a lot of the replay functionality there instead with the goal of supporting less C++ total as it seemed that we (Ouster ourselves) were the main consumers of the C++ ouster_pcap library.

But it's useful for us to know how people are/were using `ouster_pcap` -- don't think our plan there is completely clear. The more we know, the more we can maintain something that people find useful. Sorry about breaking your application on update. Think the solution you have will work for now, and we'll keep this issue in mind as we plan farther.

bk-wilson, 2022-11-24:
Thanks @kairenw 

Yes we are exclusively using the c++ SDK.  We are targeting a embedded arm platform so cross compile support is also something important to us.

Regards,

Brian