ouster-lidar/ouster-sdk #464: Legacy UDP ranges are all 0

whereskenneth, 2022-11-28:
Hello,

I am trying to use the ouster_sdk to parse some legacy UDP data. Unfortunately, all of the ranges are coming back as 0. Can you look into this? It can be reproduced using data within the SDK.

Steps to reproduce:

1.) Build SDK examples.
2.) Print all ranges (see commit in https://github.com/whereskenneth/ouster_example/blob/master/examples/representations_example.cpp for how this is being done.)
3.) Run `./representations_example ../tests/pcap/OS-2-32-U0_v2.0.0_1024x10.pcap ../tests/pcap/OS-2-32-U0_v2.0.0_1024x10.json`.

All ranges, and the sum of the ranges, are 0.

[image]

It is possible I am misunderstanding something, or calling the wrong functions to get the range data back. 

Thanks.

kairenw, 2022-12-01:
Hey @whereskenneth,

If your metadata is pretty old,  like the one that you're testing against, it's probably missing the `udp_port_lidar` in the metadata itself vs some of the [newer ](https://github.com/ouster-lidar/ouster_example/blob/master/tests/metadata/2_4_0_os-992146000760-128.json) [ones](https://github.com/ouster-lidar/ouster_example/blob/master/tests/metadata/2_4_0_os-992146000760-128_legacy.json) where whether or not it's the legacy metadata format, you can see there is a `udp_port_lidar` value. 

When we wrote `representations_example` we wanted to show the most robust way to do it against newer data, but against older data like this I'd remove [this check](https://github.com/ouster-lidar/ouster_example/blob/master/examples/helpers.cpp#L38) since they won't match as `info.udp_port_lidar` is populated from the metadata itself.

Let me know if that's your issue. If not, it's something else.

Also lol I just got a [terrible typo](https://github.com/ouster-lidar/ouster_example/blob/master/examples/representations_example.cpp#L157) in `representations_example` because of this ask, so thank you!

whereskenneth, 2022-12-02:
That indeed was the issue. Thanks. 

In case anyone else runs into this, the conditional in question that check the udp_port_lidar member is here:

https://github.com/ouster-lidar/ouster_example/blob/279c9fb36c7e217d76bdde56bc82f833fe14783b/examples/helpers.cpp#L38

closed 2022-12-02 by whereskenneth