ouster-lidar/ouster-sdk #468: The way TCP commands are deprecated actually breaks old code

peci1, 2022-12-13:
In FW 2.4, TCP command `set_config_param auto_start_flag 1` has been deprecated.

However, I don't like the way the deprecation is done. If the command succeeds, it should output `set_config_param`. However, after the deprecation, it instead outputs `error: parameter 'auto_start_flag' has been deprecated.`. It thus breaks older code that expects the only correct answer if the command has actually worked, as for example here in an older version of the driver:

https://github.com/ouster-lidar/ouster_example/blob/43107a1d14b8cd22378a830bd07aa7acca1269a4/ouster_client/src/client.cpp#L528-L529

I would suggest not touching the output of the commands until they are actually removed from the API and doing the deprecation only in documentation. Would that be possible for future releases?

kairenw, 2022-12-14:
Hi @peci1,

Do you mean the deprecation on the sensor FW and the corresponding output? `auto_start_flag` and `udp_ip` were deprecated in Nov 2020 in the comments of the [FW changelog](https://static.ouster.dev/sensor-docs/image_route1/image_route2/common_sections/API/api-changelog.html), and removed approximately 1.5 years later from the sensor API in FW 2.4. I think a PCN was sent out at the time, but it's possible it missed your team? 

There's an email list for the PCNs -- if you contact our sensor support they can make sure you get on that.

kairenw, 2022-12-14:
I can see that the wording is confusing in the FW changelog though. Would it be better if we wrote "previously deprecated parameter `udp_ip` has been removed" in the changelog? I can ask our team to change that.

peci1, 2022-12-14:
> I can see that the wording is confusing in the FW changelog though. Would it be better if we wrote "previously deprecated parameter `udp_ip` has been removed" in the changelog? I can ask our team to change that.

Exactly that. I got confused by the wording of the individual items and ignored the "Removed" section heading. Now it makes sense. If you can suggest improvement of the wording in the changelog, please, do. Thanks!

closed 2022-12-14 by peci1