ouster-lidar/ouster-sdk #471: ouster.client.core.ClientError: Failed to collect metadata

mites-123, 2022-12-15:
I am running samples example. 
**ouster_example/python/src/ouster/sdk/examples/client.py** using command **python3 client.py hostname plot-xyz-points**

I am getting this error.
The output of above command is :

**example: plot-xyz-points
sensor config of os-992133000107:
{
    "azimuth_window": 
    [
        0,
        360000
    ],
    "lidar_mode": "1024x10",
    "multipurpose_io_mode": "OFF",
    "nmea_baud_rate": "BAUD_9600",
    "nmea_ignore_valid_char": 0,
    "nmea_in_polarity": "ACTIVE_HIGH",
    "nmea_leap_seconds": 0,
    "operating_mode": "NORMAL",
    "phase_lock_enable": false,
    "phase_lock_offset": 0,
    "signal_multiplier": 1,
    "sync_pulse_in_polarity": "ACTIVE_HIGH",
    "sync_pulse_out_angle": 360,
    "sync_pulse_out_frequency": 1,
    "sync_pulse_out_polarity": "ACTIVE_HIGH",
    "sync_pulse_out_pulse_width": 10,
    "timestamp_mode": "TIME_FROM_INTERNAL_OSC",
    "udp_dest": "192.168.1.25",
    "udp_port_imu": 7503,
    "udp_port_lidar": 7502
}
Traceback (most recent call last):
  File "client.py", line 319, in <module>
    main()
  File "client.py", line 315, in main
    example(args.hostname)  # type: ignore
  File "client.py", line 209, in plot_xyz_points
    metadata, sample = client.Scans.sample(hostname, 1, lidar_port)
  File "/usr/local/lib/python3.6/dist-packages/ouster/client/core.py", line 446, in sample
    with closing(Sensor(hostname, metadata=metadata)) as sensor:
  File "/usr/local/lib/python3.6/dist-packages/ouster/client/core.py", line 166, in __init__
    self._fetch_metadata()
  File "/usr/local/lib/python3.6/dist-packages/ouster/client/core.py", line 180, in _fetch_metadata
    raise ClientError("Failed to collect metadata")
ouster.client.core.ClientError: Failed to collect metadata**

I am attaching diganostic bag file. Could you please look into it and guide me how to solve this?

From the dashboard, I tried "Reset Configuration" and checked Diagnostic latest status. It shows Operating mode as Normal and status as Initializing. Again I tried running client.py but getting same error.

Thanks in advance.

mites-123, 2022-12-15:
[image]

I have changed extension. Please change extension back to .bin

kairenw, 2022-12-15:
Hi @mites-123, 

I'm fairly sure that `plot-xyz-points` from `client.py` doesn't output the sensor config. Is that something you inserted yourself?  

I take it `hostname` has been set to the actual hostname of the sensor. Can you verify that it's been set correctly?
Also, which FW is your sensor running?

mites-123, 2022-12-16:
Yea, I have made some customization in that code.

But, I ran "fetch-metadata"  example ( with no customization ) in that I am getting this error:

example: fetch-metadata
Traceback (most recent call last):
  File "client.py", line 319, in <module>
    main()
  File "client.py", line 315, in main
    example(args.hostname)  # type: ignore
  File "client.py", line 97, in fetch_metadata
    with closing(client.Sensor(hostname, 7502, 7503)) as source:
  File "/usr/local/lib/python3.6/dist-packages/ouster/client/core.py", line 166, in __init__
    self._fetch_metadata()
  File "/usr/local/lib/python3.6/dist-packages/ouster/client/core.py", line 178, in _fetch_metadata
    legacy=self._legacy_format)
RuntimeError: Cannot initialize with sensor status: ERROR. Please ensure operating mode is set to NORMAL

Here is the system information: 

boot_id
    b523225f-2aa3-4a89-9ff5-37ccdcfbf045
ethaddr
    bc:0f:a7:00:2c:60
fw
    ousteros-image-prod-aries-v2.1.2+20210727223313.patch-v2.1.2
hostname
    os-992133000107
keeper
    v2.1.2
pn
    840-102146-D
sn
    992133000107 

Device is OS2-32.

Please guide on this. 

Thanks in advance.

kairenw, 2022-12-16:
It seems your sensor is in some sort of ongoing ERROR status. Please contact [our sensor support ](https://ouster.atlassian.net/servicedesk/customer/portal/8/group/22/create/86)directly. You can link them to this issue. I'll keep this open till it's resolved.

mites-123, 2022-12-16:
Thanks @kairenw I have raised Sensor request on support and posted link to Github issue. 
Thank you again.

kairenw, 2022-12-20:
Hi @mites-123, 

I've seen your ticket in the internal queue. Our support team will get to it soon!

mites-123, 2022-12-21:
Thanks @kairenw .

kairenw, 2023-01-12:
hey @mites-123, was your issue addressed?

closed 2023-02-09 by kairenw