ouster-lidar/ouster-sdk #473: height of depth images

mehdizayene, 2022-12-21:
Hello,

I am extracting depth images from Ouster OS1-64 and the height of the images is so small (64 pixels by 1024 for the depth). is there any way to change this (i.e having higher dimension)?

Thank you,
Mehdi

Samahu, 2022-12-21:
You can get sensors models that have the 128 laser beams instead of 64. You
can also configure the sensor with higher LiDAR modes like 2048x10  if the
sensor is capable. Please reach to our sales:
https://ouster.com/talk-to-sales/ for more information.

mehdizayene, 2022-12-24:
Hello, thank you for the reply. I actually don't see why the height of the range images is related to the number of beams?  Isn't there any way for better XYZ projections of range images in higher dimension image than 64 pixels ?

Thank you

mehdizayene, 2022-12-24:
Also, is there a way to know how do you do these 2D projections to obtain the range image ? I assume you first go from the cloud points to generate the range images ?

Samahu, 2022-12-28:
> Hello, thank you for the reply. I actually don't see why the height of the range images is related to the number of beams? Isn't there any way for better XYZ projections of range images in higher dimension image than 64 pixels ?
> 
> Thank you

I didn't quite understand your question but the the number of laser beams determines the height of generated range image. That's how spinner LIDARs work. There is no other way unless you to scale up the image using some sort of interpolation which I wouldn't recommend.

> Also, is there a way to know how do you do these 2D projections to obtain the range image ? I assume you first go from the cloud points to generate the range images ?

The sensor captures laser measurements as 2D images which you can later convert to 3D point clouds (i.e. project them) using one of the provided cartesian methods.

Hope this helps
Ussama

closed 2022-12-28 by Samahu