ouster-lidar/ouster-sdk #475: converting bag file to pcap

JJuhyeon-Kim, 2023-01-11:
Hi, thank you for your hard work.

I have one simple question.
Is there any way to convert a bag file to pcap?
I acquired my data as bag format, but I want to utilize the ouster python SDK.
I would appreciate it if you could tell me how to convert bag file to pcap file.

kairenw, 2023-01-11:
Hi @JJuhyeon-Kim what type of messages are in your bag file?

JJuhyeon-Kim, 2023-01-11:
> Hi @JJuhyeon-Kim what type of messages are in your bag file?

Thank you for your quick response!
My bag file has all kinds of messages including os_cloud_node/points and os_node/lidar packet.
Actually, I am trying to use ouster images using ouster python sdk.
So I tried to convert my bag file refering to this github code https://github.com/velodynevls/velodyne_vls.
But I failed to use ouster images (intensity ambient range) from my converted pcap file using ouster sdk.

I would appreciate it if you could me some comments!

kairenw, 2023-01-11:
So if your bag has `/lidar_packet` and `/imu_packet` it may be possible to turn it into a pcap using our tools. Mind contacting us through customer support [here](https://ouster.atlassian.net/servicedesk/customer/portal/8/group/22/create/86) ? We can try converting it for you just this once.

JJuhyeon-Kim, 2023-01-12:
> So if your bag has `/lidar_packet` and `/imu_packet` it may be possible to turn it into a pcap using our tools. Mind contacting us through customer support [here](https://ouster.atlassian.net/servicedesk/customer/portal/8/group/22/create/86) ? We can try converting it for you just this once.

Thank you for your response again!
Isn't there any open source code I can use for converting bag file?
I think there will be a lot of files to convert from now on, so I want to try it myself.

Thanks!

kairenw, 2023-01-12:
No open source code right now. Doesn't mean we can't write you something up but I'd like to get a look at what you have first.

JJuhyeon-Kim, 2023-01-12:
> No open source code right now. Doesn't mean we can't write you something up but I'd like to get a look at what you have first.

Okay, I got it. 
thank you for nice response!

kairenw, 2023-01-12:
Awesome. I will look for the support ticket on the other side.

kairenw, 2023-01-13:
Reopening so I don't forget to check back in next week.

kairenw, 2023-02-17:
Hi @JJuhyeon-Kim was your issue resolved?

JJuhyeon-Kim, 2023-02-20:
> Hi @JJuhyeon-Kim was your issue resolved?

Hi @kairenw, I could not solve the issue.
So, I just gave up using pcap files..
Thank you for asking me again!

kairenw, 2023-02-21:
What topics are in your bag? Are they `/imu_packet` and `/lidar_packet`?