ouster-lidar/ouster-sdk #478: Record multiple sensor data simulteneously

nazimshoikot1, 2023-01-19:
Hello, 

I have two sensors attached my device with different subnet masks. I can view each stream using OusterStudio separately but cannot play them simultaneously with two instances of OusterStudio.

I am trying to record the data of the two sensors simultaneously. For now, it is fine if I store it in 2 different PCAP files (and corresponding json files) as I plan to later merge the point clouds. 

I can record data from either sensor seamlessly using code similar to: https://static.ouster.dev/sdk-docs/python/examples/record-stream.html

However, it seems that I cannot have two client sensors open at the same time even if they are in different threads or processes. I get the error "ouster.client.core.ClientTimeout: No packets received within 1.0s". 

I have tried to write a script where I open a connection to sensor, store the packets in array, close it, open another sensor and store its packets in a separate array. I repeat this for a certain amount of time after which I write the packets using pcap.record and the respective arrays as the source. This approach technically works but the gap in the packets is too much for the data to be useful.

Is there a way to reliably record data from both scans simultaneously? 

Thanks in advance.

kairenw, 2023-01-20:
Hello @nazimshoikot1,

Is there any chance you're using the same ports (7502, 7503) for both sensors? And then when you start the second client the first one times out?

If you set them separately, that should fix the timeout issue. However, there's still multisensor recording which we don't support with Python right now (you can try once you fix the ports but I think that you may end up with issues with the GIL/threading - when I find some time, I will test this out. Ping me next week if I haven't responded by then).

If you're using IPv4, go ahead and just use tcpdump and specify which ports you want to listen to. That ought to be able to be read with our Python SDK.

nazimshoikot1, 2023-02-06:
My problem indeed was that I was using the same ports for both sensors. Setting different ports resolved the issue. Thank you so much!

closed 2023-02-06 by nazimshoikot1