ouster-lidar/ouster-sdk #481: how to turn off SDK warnings and info?

Shibaditya99, 2023-01-31:
`    [2023-01-31 10:40:03.347] [ouster::sensor] [info] initializing sensor: 192.168.3.2 with ports: 7502/7503\n
    [2023-01-31 10:40:04.484] [ouster::sensor] [warning] The SDK will soon output the non-legacy metadata format by default.  If you 
    parse the metadata directly instead of using the SDK (which will continue to read both legacy and non-legacy formats)", " please be 
    advised that on the next release you will either have to update your parsing or specify legacy_format = true to the get_metadata 
    function.`

how to turn off them?

Samahu, 2023-01-31:
@Shibaditya99 you can call `sensor::init_logger("error")` this would limit the log messages to error only, both `info` and `warning` messages will be filtered out.

Samahu, 2023-02-01:
@Shibaditya99 I hope my response answered your question, if not please let us know. Marking the issue as resolved.

closed 2023-02-01 by Samahu