ouster-lidar/ouster-sdk #482: Trouble compiling the C++ sdk source

Perceval62, 2023-02-01:
# What is going on
I followed the instructions on the documentation to build the examples.
I got to make a directory, cmake .. and then the cmake --build command fails with the given error.

[image]

Am I doing it right ?
am I supposed to compile the sdk and then use it ?

Samahu, 2023-02-01:
what is your build structure is like?

Did you follow the build instructions as stated here https://static.ouster.dev/sdk-docs/cpp/building.html:
```bash
mkdir build
cd build
cmake  <path to ouster_example> -DCMAKE_BUILD_TYPE=Release -DBUILD_EXAMPLES=ON
cmake --build .
```

Samahu, 2023-02-01:
I am not sure but I found that you build path has a space `ouster python` .. certain tools don't like that.

Perceval62, 2023-02-01:
Oh, you're right. I'll try renaming the folder then.
But apart from that, I followed the instruction as stated in the document that you linked.
The build structure is a verbatim clone of the master branch.

Samahu, 2023-02-02:
Closing the issue, let us know if you are still having build issues.

closed 2023-02-02 by Samahu

Perceval62, 2023-02-02:
Fair enough,
I won't be near the lidar unit until tomorrow.
I'll poke y'all to let you know if removing the space is a fix