ouster-lidar/ouster-sdk #484: Information about firing sequnce time and pattern

kolja183, 2023-02-07:
Hi, I am looking for information about the firing sequence of the os1 64 Channel LiDAR.

In other manuals about LiDAR there are information like this:

[image]
[image]

Basically telling me that the LiDAR fires 8 laserbeams at once, the pattern and direction of each laserbeam and that i takes 2.6μs for each laser to fire and collect the return energy. 

I calculated the os1 has 1000000/(20*1024) = 48.83μs to fire the 64 laserbeams at 20Hz. Is there any information aviable how many laserbeams get fired at the same time and in which pattern like in the example above? I didnt find anything in the software or hardware manual but there are many informations spread online so i was curious if you could tell me anything about the topic. 

Thank you

closed 2023-02-07 by kairenw

kolja183, 2023-02-09:
Thank you for your answer and the interesting read @kairenw.

I would just like to clarify one thing, so I dont misunderstand you.

> fired at once, at a single azimuth, before it moves to the next azimuth and fires again.

Here you mean that the angle that all lasers rotate after they fired is the same correct? The total angle of all lasers firing at once are not the same, since that would mean that there is no staggering image like explained [here](https://static.ouster.dev/sdk-docs/reference/lidar-scan.html#staggering-and-destaggering), if I understood the concept correctly. Since it would mean that each column of a lidar scan represents the same timestamp and same "total" angle and therefore there would be no need to destagger a scan.

Regards,
Kolja

kairenw, 2023-02-09:
Hi @kolja183, when I say azimuth I mean that there is a single azimuth corresponding to the position of the motor. The beams have some staggering with respect to that. You should destagger the scan in order to reposition the pixels so that a column represents a single azimuth as opposed to a timestamp.

stockerd, 2023-02-09:
Although I'm not an Ouster representative, I can clarify regarding a question posed by @kolja183:  The staggering in azimuth angles is not due to the timing of the firing, but rather due to small angular differences in the scanner optics at the silicon level.  Those angle offsets are measured during manufacturing and reported to the user for destaggering.