ouster-lidar/ouster-sdk #487: Visualize a running ouster sensor using open3d in python

marj44, 2023-02-27:
Hello, 

I'm trying to live stream real time data from my Ouster sensor (OSO-128-U) using Open3d in python.

I haven't been able to find a lot of [documentation](https://static.ouster.dev/sdk-docs/python/examples/visualizations.html#visualization-with-open3d) on how to do this. 

The steps I have followed are:
1. Configure my sensor: 
python -m ouster.sdk.examples.client os-123456789123 configure-sensor
2. Run my code:
' `from ouster import  client` '
' `import open3d as o3d` '
' `sensor = client.Sensor('os-123456789123', 7502, 7503) ` '
' `# Create a visualizer object` '
' `vis = o3d.visualization.Visualizer() ` '
' `vis.create_window()` '
' `# Create a point cloud object` '
' `pcd = o3d.geometry.PointCloud()` '
' `vis.add_geometry(pcd)` '
' `# Start streaming data from the Ouster device` '
' `with client.run_forever() as gen:` '
' `    # Loop through each packet in the stream` '
' `    for packet in gen():` '
' `        # Check if the packet is a lidar packet` '
 ' `       if packet['type'] == client.PACKET_LIDAR:` '
' `            # Get the lidar data from the packet` '
 ' `           lidar_data = packet['data']` '      
' `            # Convert the lidar data to a numpy array` '
' `            lidar_array = client.utils.parse_lidar_packet(lidar_data)` '
' `            # Convert the lidar data to a point cloud` '
' `            pcd.points = o3d.utility.Vector3dVector(lidar_array.T[:, :3])` '
' `            # Update the visualizer` '
' `            vis.update_geometry(pcd)` '
' `            vis.poll_events()` '
' `             vis.update_renderer() >   ` '

But I get the following error with a white screen:
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: module 'ouster.client' has no attribute 'run_forever'

I have checked the [documentation](https://static.ouster.dev/sdk-docs/python/examples/visualizations.html#visualization-with-open3d) and found a single line that replays pcap files but not to visualize a running sensor using open3d.
py-3-mouster.sdk.examples.open3d^--pcap$SAMPLE_DATA_PCAP_PATH--meta$SAMPLE_DATA_JSON_PATH--start84--pause

Thank you and if there is a better way to do what I'm trying to do would be greatly appreciated.

kairenw, 2023-02-27:
Hi @marj44,

So for  your code itself -- there simply is no `run_forever` attribute within the Ouster client, hence your error.

If you want to visualize from a sensor using our sample open3d visualization code, you can use:
`python3 -m ouster.sdk.examples.open3d --sensor $SENSOR_HOSTNAME` 

The code that runs it can be seen here: https://static.ouster.dev/sdk-docs/_modules/ouster/sdk/examples/open3d.html

You can see that the code uses the `Scans` interface to batch packets and deals with the data in frames. What are you trying to do aside from simply visualize from a running sensor?

marj44, 2023-02-28:
Thank you for the input! I think I got mixed up somehow. 

I would ultimately like to do something similar to this [sample problem](https://ouster.com/blog/object-detection-and-tracking-using-deep-learning-and-ouster-python-sdk/) from Ouster but using open3d and point cloud data. 

I see from the example that they are using sample lidar data converted to grayscale images but I would like to do it using the live stream point cloud data. I understand I will have to save point clouds (.pcd/.ply files) and train an algorithm on those files. I'm not familiar with point cloud object detection with ouster. The sample problem doesn't discuss point cloud object detection so any resources would be appreciated.

kairenw, 2023-02-28:
Hmm that's a little beyond what we support on this repo. There are a variety of methods for object detection, and it's best to start with an idea of how you want to approach the problem.

The approach followed in the blog post "Object Detection and Tracking using Deep Learning and Ouster Python SDK" is applying YOLOv5, a 2D CNN, to the 2D range image, which has a 1:1 correspondence to the 3D point cloud. That's one way to do object detection on the point cloud.  I think with a little effort it should be possible to expand the code that Fisher posted to work with a sensor instead of against pcaps, thus fulfilling your goal.

You could also go for object detection directly on the 3D point cloud, and within this there are also a variety of approaches. Most papers should be accompanied with Github repos that you should be able to clone and try. A good place to start if you're doing urban settings might be the Kitti 3d Object detection evaluation. At that point, it's just a question of adjusting it to work with your sensor or changing the format of the data into one that is consumed by the code you need.

Closing this issue now.

closed 2023-02-28 by kairenw