ouster-lidar/ouster-sdk #488: How to get intensity and elongation values?

irmuun20, 2023-02-27:
Hello, I want to test [centerpoint](https://github.com/tianweiy/CenterPoint) 3D object detection model on data collected via ouster lidar. An ouster lidar returns following:

- `Range` - Range in millimeters, discretized to the nearest 1 millimeters with a maximum range of 524m. Note that range value will be set to 0 if out of range or if no detection is made.
- `Calibrated Reflectivity` - Sensor Signal Photons measurements are scaled based on measured range and sensor sensitivity at that range, providing an indication of target reflectivity.
- `Signal Photons` - Signal intensity photons in the signal return measurement are reported.
- `Near Infrared Photons` - NIR photons related to natural environmental illumination are reported.

The model, I am using, was trained on [Waymo](https://www.waymo.com/open) dataset which consists of following:
`x, y, z, intensity, elongation`

- `Intensity` - A measurement indicating the return strength of the laser pulse that generated the LiDAR point, partly based on the reflectivity of the object struck by the laser pulse. 
- `Elongation` - The elongation of the laser pulse beyond- its nominal width. Elongation in conjunction with intensity is useful for classifying spurious objects, such as dust, fog, rain. Our experiments suggest that a highly elongated low-intensity return is a strong indicator for a spurious object, while low intensity alone is not a sufficient signal.

I can produce x, y, z coordinates from `Range` value but I don't know how to get `intensity` and `elongation` from ouster lidar. I am not sure if `Signal Photons` from ouster lidar is the same as [intensity](https://openaccess.thecvf.com/content_CVPR_2020/papers/Sun_Scalability_in_Perception_for_Autonomous_Driving_Waymo_Open_Dataset_CVPR_2020_paper.pdf) in Waymo dataset.

I'd really appreciate it if someone will help me to get these two values from ouster lidar. Thanks in advance.

kairenw, 2023-02-27:
Hi @irmuun20,

Do you know the units of the intensity measurement from Waymo? This issue [here](https://github.com/waymo-research/waymo-open-dataset/issues/93) seems to indicate that it's raw photons? Is that true?

For elongation, I don't believe there is anything equivalent that we report from our sensor.

irmuun20, 2023-03-08:
Yes it is raw intensity value that can go up to 6000

kairenw, 2023-03-08:
Great - I'd suggest using `SIGNAL` for the intensity, though you may find you want to scale it to be able to use your trained model.

irmuun20, 2023-03-08:
Yes that is the problem. when I scale it like following, it shows misaligned results:
`intensity = tanh(SIGNAL / 255.0)`
I wonder what is the correct way to normalize it.
[image]

irmuun20, 2023-03-08:
When I use tanh(SIGNAL), the minimum value is 0.99.
[image]

irmuun20, 2023-03-09:
https://github.com/HaohaoNJU/CenterPoint/issues/29#issuecomment-1461186498