ouster-lidar/ouster-sdk #490: Synchronize two scans from different PCAP files with timestamp

nazimshoikot1, 2023-03-02:
Hello,

I have 2 LiDARs that I use to collect two different PCAP files. I would like to synchronize one frame from one PCAP to the corresponding frame in the other PCAP file according to the same (or closest) timestamp. 

What would be the best way to get the timestamp of a frame in either standard formats such as UTC or relative to the same point in time? 

When I open the PCAP in ouster studio and play it, I see the time updating on the bottom right corner like this:

[image]

I see the same time range in the two PCAP files so I'm guessing this is close to what I am looking for, but I have not yet been able to find a way to get this time of the scan from the python ouster-sdk. 

Any help with how to get this time, or to get the actual UTC time from timestamp would be appreciated. Thanks!

Samahu, 2023-03-02:
You definitely need to have a PTP master clock that both of your lidars synchronize with. You may follow the [Ouster PTP setup guide](https://static.ouster.dev/sensor-docs/image_route1/image_route2/appendix/ptp-quickstart.html) here on how to achieve that.

With regard to converting your time to UTC time -which is only necessary if you need your time in UTC - check my answer on the topic [here](https://github.com/ouster-lidar/ouster_example/issues/486)

closed 2023-03-02 by Samahu

nazimshoikot1, 2023-03-03:
Thank you for the PTP setup link. I'll make sure to set it up for before data collection next time. 

However, I also need to synchronize the the frames on the data that I collected for now since due to some limitations, it will be some time before I can collect the data again. I see that the time (1677553572.93 in the screenshot) is available on both the pcaps, so I'm thinking I can use that to compare the frame times. However, the measurement times for the packets (2248723738690, for instance) are values that are quite different. Not sure if I am missing something obvious, but could you tell me how I could get the value 1677553572.93 for Frame 14 (as shown above in OusterStudio) using Ouster-SDK

Samahu, 2023-03-03:
If you didn't setup a shared PTP master for the two sensors then every sensor is time stamping the packets using its own clock; the sensor clock starts ticking from the time you connected the sensor to power. Even if you powered up the two sensors at the same time (which I doubt you did), without a PTP master the clocks of the two sensors will eventually become out of sync (maybe a tiny bit). Because of that the timestamps that you see in the pcaps even if it looks similar or same that means nothing. You need a PTP master for the two sensors.

nazimshoikot1, 2023-03-06:
For my setup, they are indeed turned on at the same time. I understand that they will still be slightly out of sync and I have to account for that but just until I get it fixed for next time, could you tell me where exactly the time for the frame (1677553572.93 in this case is stored) and how I can get that value using ouster-sdk?

Samahu, 2023-03-06:
The sensor itself doesn't generate a frame time but has a timestamp for every column. Typically, people use the first column of the frame as the timestamp of the entire frame. You can track the beginning of the frame based on the `frame_id` and `measurement_id` fields of the packets. You need to account for [potentially missing packets in the frame which could affect the frame time](https://github.com/ouster-lidar/ouster-ros/issues/47).