ouster-lidar/ouster-sdk #494: How do I determine the correct ChanField type?

themightyoarfish, 2023-03-13:
**Describe your question**

We recently got our first Rev7 sensor, and it seems to behave differently in some way. We are not setting `udp_profile_lidar` when filling the config, and I'm not sure what it defaults to, but with the current master branch of `ouster_example` installed, I now get `std::invalid_argument: Accessed field at wrong type` when trying to access the `REFLECTIVITY` channel as

```
    const auto& reflectivity = ls.field(ouster::sensor::REFLECTIVITY);
```

Now looking at the code, this appears to default the type to `uint32_t`, so maybe this changes in recent firmware versions? I can see a changelog entry for 2.3.0 or so, but our other sensors were using the same the same client code with 2.4.x, so I'm confused what the problem is.

```
    const auto& reflectivity = ls.field<uint8_t>(ouster::sensor::REFLECTIVITY);
```

exhibits the same result.

When I instead set `udp_profile_lidar` to `ouster::sensor::UDPProfileLidar::PROFILE_RNG15_RFL8_NIR8`, I would expect – as per `parsing.cpp` – the `lb_field_info` struct to be used, which states that reflectivity has type `uint8`. But it only works when I access as `uint16_t`.

How can I understand the influence of the Lidar profile on the channel field types?

**Screenshots**
n/a

**Platform (please complete the following information):**
- Ouster Sensor? OS1-124
- Ouster Firmware Version? 3.0.1
- Programming Language? C++
  - For Python include the targeted Python version
- Operating System? MacOS
- Machine Architecture? arm
- git commit hash (if not the latest). 279c9fb36c7e217d76bdde56bc82f833fe14783b

themightyoarfish, 2023-03-13:
[In this document](https://static.ouster.dev/sensor-docs/image_route1/image_route2/sensor_data/sensor-data.html#configurable-data-packet-format) it also states that reflectivity for this profile should be 8 bits, so I'm confused what I am doing wrong.

kairenw, 2023-03-13:
Hey @themightyoarfish,

Yeah, it's a little confusing. You're not doing anything wrong! I assume that since you didn't use to set anything, you used to default to LEGACY (which is how everything worked) and now the default on FW 3.0 has changed to the SINGLE RETURNS profile, which is why you're getting the bits error. Let me explain a little.

The types of fields in the LidarScan do not necessarily reflect those from the sensor (which is the documentation you linked). For example, for the low bandwidth profile, the RNG15 might imply using 16 bits in the client to capture - but then that would be the  unscaled version, so for ease of use with the client, we've chosen to convert that into the same units as the other ranges (in mm) and use a `uint32_t` always.

In the case of REFLECTIVITY, there was a time we believed that 16 bits would come off the sensor in non-low-bandwidth profiles, so we started at 16 bits across all profiles instead of 8 bits (since the low bandwidth would then be scaled up). REFLECTIVITY ended up being 8 bits, but in order to keep the churn to a minimum, we decided not to upgrade it to 8 bits ([as we did at one point consider](https://github.com/ouster-lidar/ouster_example/issues/398#issuecomment-1173515974).

In any case, you can see the expected size specified for default on the single return profile here: https://github.com/ouster-lidar/ouster_example/blob/c78d7927658e3c0de263df26a0db51a26766bdcb/ouster_client/src/lidar_scan.cpp#L73

Hope that makes it a bit more clear!

themightyoarfish, 2023-03-13:
Ok this is helpful. I would suggest adding some notes to the documentation on this topic.

iAmir97, 2023-05-15:
Following up on this, I am a beginner developer trying to create a program that is able to support both legacy and single data return profiles but using 

```cpp 
const auto& reflectivity = ls.field<uint16_t>(ouster::sensor::REFLECTIVITY);
``` 
will fail upon using the legacy profile, can you suggest a way I can parse the reflectivity or any other channel that won't fail upon using either profile? can suggest a way to manipulate the ouster API to access the fields or suggest a way to make it different on runtime, thank you for your help!

Samahu, 2023-05-15:
@iAmir97 you would need to use the `ls.field_type` method to verify the underlying type/size of the field then perform the proper cast.

giulionf, 2024-10-16:
Seems like its back at uint8_t?

Samahu, 2024-10-16:
Yes, `REFLECTIVITY` was changed back to `uint8_t` as part of the [last release](https://github.com/ouster-lidar/ouster-sdk/releases/tag/release-0.13.0#:~:text=%5BBREAKING%5D%20changed%20REFLECTIVITY%20channel%20field%20size%20to%208%20bits.%20(Important%20%2D%20this%20makes%20the%20SDK%20incompatible%20with%20FW%202.0%20and%202.1.)). Your sensor needs to be on FW 2.2 or later due to this change.

closed 2024-10-16 by Samahu