ouster-lidar/ouster-sdk #499: I want more examples for CV tasks

FlyAIDrone22, 2023-03-21:
**Describe your question**
A clear and concise description of your question.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Platform (please complete the following information):**
- Ouster Sensor? \[e.g. OS-0, OS-1, ..\]
- Ouster Firmware Version? \[e.g. 2.3, 2.4, ..\]
- Programming Language? \[e.g. C++, Python, ..\]
  - For Python include the targeted Python version
- Operating System? \[e.g. Linux, Windows, MacOS\]
- Machine Architecture? \[e.g. x64, arm\]
- git commit hash (if not the latest).

FlyAIDrone22, 2023-03-21:
I want more examples for CV tasks as i using lidar 32

kairenw, 2023-03-21:
thanks @FlyAIDrone22 ! We'll leave this open until we can fulfill it. No promises on timeline