ouster-lidar/ouster-sdk #502: How can I compute the (relative) timestamp of each point?

themightyoarfish, 2023-03-29:
**Describe your question**

Each measurement block appears to have a timestamp as documented [here](https://static.ouster.dev/sensor-docs/image_route1/image_route3/sensor_data/sensor-data.html#lidar-data-packet-format), but I couldn't quickly find how points are associated to blocks exactly.

What arithmetic do I have to perform to get the timestamp for each point? There is code [here](https://github.com/ouster-lidar/ouster_example/blob/13ea8e8b8a4951fb630dbc9108666995c8443bf6/ouster_ros/src/ros.cpp?_pjax=%23js-repo-pjax-container%2C%20div%5Bitemtype%3D%22http%3A%2F%2Fschema.org%2FSoftwareSourceCode%22%5D%20main%2C%20%5Bdata-pjax-container%5D#L126) which simply accesses the `LidarScan` header for each column of the data, does this mean that each block contains one column? And is this the definitive way to do it? This wasn't obvious to me from when this question got [asked previously](https://github.com/ouster-lidar/ouster_example/issues/184).

JoeNajm, 2023-04-01:
Hello @themightyoarfish , did you find a solution ? I have the same issue :(

kairenw, 2023-05-30:
Hi @themightyoarfish,

Are you working mostly with the ROS code ? Your links point mostly there - you can trust that the timestamp provided in the /points topics are correct for each point.

The Ouster sensor outputs one timestamp per column - see the illustration of the packet format [here](https://static.ouster.dev/sensor-docs/image_route1/image_route3/sensor_data/sensor-data.html#lidar-data-packet-format). That timestamp is recorded into the `timestamp()` header, so the `i`th element is the timestamp associated with the `i`th column. The thing to remember is that these are the timestamps associated with the staggered columns - which are the default when you access the `img_t` returned from `scan.field()`. Every point in that column will have the same timestamp.

Does that help a bit?

twslankard, 2023-09-06:
I'm closing this issue due to inactivity. Please open a new issue for SDK-specific inquiries or contact [support@ouster.io](mailto:support@ouster.io) for device-related support.

Best Regards,

Tom
Staff Software Engineer
Ouster, Inc.

closed 2023-09-06 by twslankard

themightyoarfish, 2024-03-27:
I read in several places, that the timestamp for a point is given by `const auto ts = (ls.header(v).timestamp - scan_ts).count();`, but isn't that a relative timestamp since the start of the rotation? What I need is the absolute timestamp, I would have thought this is `scan_ts` (obtained somehow) plus the relative timestamp obtained from the header. Is this not the case? What should `scan_ts` actually be?

themightyoarfish, 2024-03-28:
Reading https://github.com/ouster-lidar/ouster-ros/blob/5a08f89e21dabfb78ae359f9bfc4ba869a275104/src/lidar_packet_handler.h it seems like quite a bunch of effort to compute a scan time stamp to use for this calculation.

Samahu, 2024-03-28:
@themightyoarfish this post gives a high level description of how timestamps are stored and how to translate between relative and absolute https://github.com/ouster-lidar/ouster-ros/discussions/206

themightyoarfish, 2024-07-29:
So do I understand correctly that there is no timestamp for each `LidarScan` and only timestamps for each point, relative to the start of the scan? And the complex logic in `ouster-ros` is necessary to map those to absolute times in the presence of a column window? I am not using ROS, so I have to get there myself somehow.

Samahu, 2024-07-29:
yeah we don't have a LidarScan time but rather a time per measurement/column. You are free to use the proper time first, mid, last but you need consider that the azimuth window setting. You can check or implement something similar to the first_valid_column_ts method in the python code. However, this doesn't tolerate missing leading packets so the time could fluctuate between frames if you have lots of dropped packet. The ouster-ros driver uses the timestamp from the first measurement as the PointCloud timestamp and interpolates the time in case some of LidarScan leading packets are missing.

themightyoarfish, 2024-11-09:
I'm still confused by this. I am now using the packet handling code from `ouster-ros` and get `scan_ts` from that, but unless that is PTP-synced, this is a system timestamp, while the measurement block timestamps are ns since sensor start. Therefore, this code here will always evaluate the timestamp to zero: 

https://github.com/ouster-lidar/ouster-ros/blob/9711abd69834972a675e4a8ffad261231b021364/src/point_cloud_compose.h#L158

I'm not sure why this is not a problem in the ROS driver.

Samahu, 2024-11-11:
In general `scan_ts` is measurement timestamp[0] as such for the relative timestamp for the first measurement `t0` would be zero. But for next one `t1` would be `1e9/(1024*10)` assuming a LidarMode of 1024x10 and t2 should be `2e9/(1024*10)` and so on. This should be the case for all frames with limited variance due to the mechanical motion.

The only exception is if you have dropped packets at the beginning (in which we use linear interpolation) or if there the clock jumps.

Hope this helps.

themightyoarfish, 2024-11-27:
The lidar scan `timestmap` in 2048x10 mode has values ranging from `3771658217864` for column 1 to `3771758155224` for column 2047. While the delta is the expected 100ms, the absolute values do not seem to make sense. Should the first column not have a timestamp of 0 ns?