ouster-lidar/ouster-sdk #506: how to add a pull-up resistor with interface box to get 3.3V

zhiranyan, 2023-04-21:
**Describe your question**
A clear and concise description of your question.

Hello, according hardware manual I need one pull-up resistor to amplify the MULTIPURPOSE_IO voltage. Now I have set this pin as OUTPUT mode. I have connected this pin and GND pin with another camera via two jumper cables. How to add the resistor in the circuit? Could you provide more details about this? 

Thanks a lot.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Platform (please complete the following information):**
- Ouster Sensor? \OS-1
- Ouster Firmware Version? \[e.g. 2.3, 2.4, ..\]
- Programming Language? \[e.g. C++, Python, ..\]
  - For Python include the targeted Python version
- Operating System? \ Ubuntu 
- Machine Architecture? \ x64
- git commit hash (if not the latest).

kairenw, 2023-04-21:
Hi @zhiranyan,

I recommend contacting [our technical customer sensor support,](https://ouster.atlassian.net/servicedesk/customer/portal/8) as this Github exists to address questions about the Ouster SDK. I'll keep this issue open for now and you can let me know if you don't hear back from our customer support.

kairenw, 2023-05-01:
Closing - feel free to reopen if our technical customer sensor support hasn't replied!

closed 2023-05-01 by kairenw