ouster-lidar/ouster-sdk #511: set_config function not working

WenTheProgrammer, 2023-05-02:
I have a OS0-128. Firmware version v2.4.0. I am using the sensor with Spyder in Windows 10. I was trying to set sensor parameters inside the program, but somehow the function `set_config()` is not working for me.

My code:
```
from ouster import client
hostname = 'os-122209001028.local'
config = client.SensorConfig()
config.operating_mode = client.OperatingMode.OPERATING_NORMAL
config.lidar_mode = client.LidarMode.MODE_2048x10
config.udp_port_lidar = 7502
config.udp_port_imu = 7503
config.signal_multiplier = 3
config.azimuth_window = (120000, 240000)
client.set_config(hostname,config,persist=True,udp_dest_auto=True,force_reinit=True)
```
The error message:
```
TypeError: set_config(): incompatible function arguments. The following argument types are supported:
    1. (hostname: str, config: ouster.client._client.SensorConfig, persist: bool = False, udp_dest_auto: bool = False) -> None

Invoked with: 'os-122209001028.local', <ouster.client._client.SensorConfig object at 0x0000025DC0048AB0>; kwargs: persist=True, udp_dest_auto=True, force_reinit=True
```
For some reason, when I remove `force_reinit=True` inside `client.set_config()`, the error message becomes:
```
ValueError: Error on 'set_config_param': error: parameter 'auto_start_flag' has been deprecated.
```
Any help is greatly appreciated!

kairenw, 2023-05-03:
Hi @WenTheProgrammer,

Which version of the Python SDK are you using? Also, can you tell me which FW you're running?

WenTheProgrammer, 2023-05-03:
Hi @kairenw,
I am using ouster-sdk 0.4.1. My firmware version is v2.4.0.

kairenw, 2023-05-04:
Great - so the problem is that you're using an old version of the SDK with a newer firmware. Try upgrading your ouster-sdk to the latest (0.8.1) with `pip install -u ouster-sdk`

WenTheProgrammer, 2023-05-04:
I have to use `pip install ouster-sdk --upgrade` to upgrade the ouster-sdk to 0.8.1. However, my code now gives another error:
```
RuntimeError: CurlClient::execute_get failed for the url: [http://os-122209001028.local/api/v1/system/firmware] with the error message: Couldn't resolve host name
```

kairenw, 2023-05-04:
Can you ping the sensor at that hostname?

WenTheProgrammer, 2023-05-04:
Yes I am able to.

kairenw, 2023-05-04:
Can you show me the output?

WenTheProgrammer, 2023-05-04:
There is really no output in Spyder. Here is what I did:

```
import os
hostname = "os-122209001028.local"
response = os.system("ping " + hostname)
```
I got a "0" response, meaning the address is up. 

I was also able to ping the address in command prompt:

[image]

kairenw, 2023-05-04:
What happens if you use the python requests library on `http://os-122209001028.local/api/v1/system/firmware` ?

WenTheProgrammer, 2023-05-04:
I am not able to ping that one this time in both Spyder and command prompt.

kairenw, 2023-05-04:
Ok - I'm going to send you to our customer support so they can get on a call with you. This is a networking issue as opposed to a bug in the SDK as of now (my guess) 

Go here: https://ouster.atlassian.net/servicedesk/customer/portal/8/group/22/create/86
And you can mention this ticket in your description

WenTheProgrammer, 2023-05-04:
I am not having much luck there. I opened a ticket two months ago, my issue is still unsolved to this day and no one is replying to my ticket, even after I escalated the ticket. This is why I'm posting my issue here.

WenTheProgrammer, 2023-05-04:
I forgot to mention that my code worked perfectly fine with the older Python SDK version and could receive data from the sensor with no problem. I wanted to configure and force reset my sensor using the `client.set_config` function, which is actually a workaround for my unsolved issue, but I received the error that I showed at the beginning of this ticket. Now after upgrading the SDK to the newest version, even my old code, which used to work fine and does not have `client.set_config`, no longer works.

kairenw, 2023-05-04:
Right - now you are experiencing connection issues which you didn't have previously. Sometimes that happens! Our support team will contact you shortly

kairenw, 2023-07-07:
Issue was addressed, I believe. Please reopen if not true.

closed 2023-07-07 by kairenw