ouster-lidar/ouster-sdk #515: Support for Nvidia Drive Orin Kit

razor123max, 2023-05-15:
**Describe your question**
The OS1-128-U is not supported out-of the box with Drive OS / **Driveworks samples on Nvidia Drive AGX ORIN kit**

**Platform (please complete the following information):**
- Ouster Sensor? **OS1-128-U**
- Programming Language? **C++**
- Operating System? **Linux**
- Machine Architecture? **ARM**

Is there any way to use it? Thank you.

Samahu, 2023-05-15:
@razor123max The NVIDIA Driver Orin kit for Ouster was supported up to JetPack 3.5. Which JetPack version are you targeting? Have you tried to reach out to the customer support?

razor123max, 2023-05-15:
@Samahu I think Jetpack is applicable only to the Jetsons platform, I need support for drive platform. Thanks.

razor123max, 2023-05-15:
Closed by mistake

Samahu, 2023-05-15:
Sorry I meant the DriveWorks version. The ouster driver is available for 3.5.

Samahu, 2023-05-17:
I'll try to get back to you on the issue by the end of this month

Samahu, 2023-08-07:
Hi @razor123max, sorry for the overdue response. We are targeting a release by the end of this quarter. You may reach our customer support for a beta version for now.

Samahu, 2024-02-17:
Hi @razor123max, happy to inform you that Ouster is now providing support of its sensors for the NVIDIA Drive platform :tada: . You can access the drivers through [Ouster Downoads](https://ouster.com/downloads) page. The binary targets OS 6.0.8.1 / DW 5.14 and it supports recent sensor Revs back to 5 and up FW 2.0 up to 3.1.

Let us know if you have any questions.

closed 2024-02-17 by Samahu