ouster-lidar/ouster-sdk #517: PCAP frame count (Python SDK)

d33dler, 2023-05-19:
**Describe your question**
How can I efficiently find the number of frames in a PCAP file (in python), assuming we are using pcap.Pcap/client.Scans objects? Currently I simply iterate over the files, but this process is slow.

**Screenshots**
N/A

**Platform (please complete the following information):**
- Ouster Sensor? OS 1
- Ouster Firmware Version? 2.4
- Programming Language? Python
  - For Python include the targeted Python version Python3.9
- Operating System? linux

d33dler, 2023-05-19:
No, that's what i'm doing now, this is slow, especially for large files.

Samahu, 2023-05-19:
You could try the frame_count method from our **IndexedPcapReader**. It is implemented in C++ and then exposed to python, I haven't tested it myself but I would suppose it should somewhat faster. Here is an example invocation of the method in one of our python tests: https://github.com/ouster-lidar/ouster_example/blob/ff46c961b690daae3e18e39999ef8651081cbc66/python/tests/test_pcap.py#L515

Hope this helps!

d33dler, 2023-05-20:
I'll try it out, thanks @Samahu !

kairenw, 2023-05-22:
We hope to do some bindings on `IndexedPcapReader` eventually, but no promises on time. How slow are we talking?

d33dler, 2023-05-31:
@kairenw Obviously it depends on the recording set (I do batch processing oftenly and it can total 24 hours of recording) and calculating the frame count can be a slow task and take up to 10 minutes, despite having it parallelised. I changed to Samahu's  solution proposed using `IndexedPcapReader` and it is somewhat faster.

closed 2023-05-31 by d33dler