ouster-lidar/ouster-sdk #521: OS1 LIDAR does not work properly under Windows/ WSL2 environments

arunumd, 2023-06-05:
**Describe the bug**
I am trying to run the example files under the examples directory of this repository. When I run the `./client_example LIDAR-IP ""` the sensor gets stuck at the point when it has to retrieve the point clouds and save them as .csv files.

**To Reproduce**
Steps to reproduce the behavior (steps below are just an example):
1. Install WSL2 on Windows10/11 pc with the default version as of today (Ubuntu 22.04.2 LTS)
2. Open WSL window and compile this repository after installing all dependencies (`sudo apt install build-essential cmake libjsoncpp-dev libeigen3-dev libcurl4-openssl-dev libtins-dev libpcap-dev libglfw3-dev libglew-dev libspdlog-dev`)
4. run the  `./client_example LIDAR-IP ""` example as the normal user (not as root)
5. Notice that the example appears to connect to the LIDAR and retrieves the sensor metadata (Step 1); but fails to retrieve the point clouds and save them as .csv files (Step 2)

**Screenshots**
[image]

**Platform (please complete the following information):**
- Ouster Sensor? OS-1-32-U0
- Ouster Firmware Version? 2.5.1
- Programming Language? C++
- Operating System? Windows 10/11 with WSL2
- Machine Architecture? x64
- git commit hash (if not the latest). ff46c961b690daae3e18e39999ef8651081cbc66

chrisbayruns, 2023-06-05:
WSL2 works under the hood as a full virtual machine. This means the network is not the standard windows networking. The default seems to be a NAT networking setup. You will need to configure the WSL2 networking to allow the sensor to communicate. https://learn.microsoft.com/en-us/windows/wsl/networking#accessing-a-wsl-2-distribution-from-your-local-area-network-lan

arunumd, 2023-06-05:
> WSL2 works under the hood as a full virtual machine. This means the network is not the standard windows networking. The default seems to be a NAT networking setup. You will need to configure the [...]

It does not seem to provide all the insights needed to get it working. The relevant command of the form `netsh interface portproxy add v4tov4 listenport=4000 listenaddress=0.0.0.0 connectport=4000 connectaddress=192.168.101.100` did not seem to have done anything. It still doesn't work. A detailed guidance of how to get a OS1 LIDAR working on a Virtualbox or WSL2 environment would be helpful. Maybe you can try to reproduce the same problem at your end and post it here when you find a solution

kairenw, 2023-06-05:
Hi @arunumd -  supporting WSL2 on Ubuntu is not something we have bandwidth for right at this moment. We can keep this ticket open for quick troubleshooting esque questions, but it's unlikely we'll get around to providing a detailed guide for it.

I'd try contacting [our customer support](https://ouster.atlassian.net/servicedesk/customer/portal/8) / your sales representative about your use case

kairenw, 2023-06-05:
I encourage you to contact our customer support as they can get on a call with you and make sure your networking configuration is correct

arunumd, 2023-06-07:
I went to the Dell website and searched up for Dell Precision 7670 drivers and downloaded and installed the latest Ethernet driver. After doing this the example code works correctly in Windows only. WSL2 still doesn't work. So at least the windows part of the problem is now solved. WSL2 is pending.

closed 2023-06-07 by arunumd

arunumd, 2023-07-21:
Update:-

WSL2 has an active issue relating to low network speeds on their [GitHub issues page of Microsoft](https://github.com/microsoft/WSL/issues/4901). The suggestion provided is to use WSL1 which offers better speeds (close to native Windows host speed). After doing this downgrade to WSL1, the scanner works. This is now just a work around to solve the issue temporarily. WSL2 is still an issue. But it is irrelevant to Ouster.