ouster-lidar/ouster-sdk #523: Make logging configurable

themightyoarfish, 2023-06-12:
**Is your feature request related to a problem? Please describe.**

Client class logs to stdout by default. The only way to configure this seems to be setting the output log file with `init_logger()`. This prevents the user from handling the log messages themselves, e.g. ingestion into the application's logging.
There's also no trivial portable way to disable logging, so inevitably, something is getting spammed.

**Describe the solution you'd like**

Make use of spdlog to allow custom loggers.

**Describe alternatives you've considered**
n/a

**Targeted Platform (please complete the following information only if applicable, otherwise dot N/A):**
- Programming Language? C++

kairenw, 2023-06-20:
Hi @themightyoarfish,

Thanks for the suggestion. To make it a bit more clear, are you suggesting that we shift [logging.h](https://github.com/ouster-lidar/ouster_example/blob/ff46c961b690daae3e18e39999ef8651081cbc66/ouster_client/src/logging.h#L1) to `include/` and exposre more ways to configure the sink for our logger from it?

themightyoarfish, 2023-06-20:
Since `spdlog` is a dependency anyway, the intuitive way to handle this in my opinion would be to let the user inject their own logger so they can do with ouster logs whatever they wish.