ouster-lidar/ouster-sdk #527: Issue building ouster-sdk on Windows 11

Nico-Trimble, 2023-06-16:
**Description**
I'm working on a python application that makes use of ouster-sdk on Windows 11, and I'm having trouble building ouster-sdk. I've made this a question rather than a bug report because I think it's something I'm doing wrong rather than a bug in ouster-sdk or any other software.

**Problem Details**
First, I tried installing the wheel through pip following the directions [here](https://static.ouster.dev/sdk-docs/installation.html):
`python3 -m pip install "ouster-sdk[examples]"`
This fails building the wheel.

Then I tried downloading the wheel for windows 10 (with the though that windows 11 should be backwards compatible). This fails:
`PS> python3 -m pip install ouster_sdk-0.8.1-cp310-cp310-win_amd64.whl
ERROR: ouster_sdk-0.8.1-cp310-cp310-win_amd64.whl is not a supported wheel on this platform.`

Now, I'm trying to build the wheel following the directions [here](https://static.ouster.dev/sdk-docs/python/devel.html).

I installed Visual Studio 2017 Build Tools, Python and vcpkg.

I ran `vcpkg install --triplet=x64-windows eigen3 jsoncpp libtins pybind11 glfw3 glad[gl-api-33] spdlog` successfuly.

I cloned the repository at [https://github.com/ouster-lidar/ouster_example](https://github.com/ouster-lidar/ouster_example) and updated the environment variables as specified. Now this is the output I'm getting. Any help would be appriciated!

`PS C:\Users\nzucca\ouster_example> python3 -m pip wheel --no-deps "$env:OUSTER_SDK_PATH\python"
Processing c:\users\nzucca\ouster_example\python
  Installing build dependencies ... done
  Getting requirements to build wheel ... done
  Preparing metadata (pyproject.toml) ... done
Building wheels for collected packages: ouster-sdk
  Building wheel for ouster-sdk (pyproject.toml) ... error
  error: subprocess-exited-with-error

  × Building wheel for ouster-sdk (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [96 lines of output]
      running bdist_wheel
      running build
      running build_py
      running build_ext
      -- Ouster SDK location: C:/Users/nzucca/ouster_example
      CMake Error at C:/Users/nzucca/AppData/Local/Temp/pip-build-env-75xmxof4/overlay/Lib/site-packages/cmake/data/share/cmake-3.21/Modules/CMakeDetermineSystem.cmake:130 (message):
        Could not find toolchain file: ./vcpkg/scripts/buildsystems/vcpkg.cmake
      Call Stack (most recent call first):
        CMakeLists.txt:11 (project)

      CMake Error at CMakeLists.txt:11 (project):
        Failed to run MSBuild command:

          C:/Program Files (x86)/Microsoft Visual Studio/2017/BuildTools/MSBuild/15.0/Bin/MSBuild.exe

        to get the value of VCTargetsPath:

          Microsoft (R) Build Engine version 15.9.21+g9802d43bc3 for .NET Framework
          Copyright (C) Microsoft Corporation. All rights reserved.

          Build started 6/16/2023 1:04:24 PM.
          Project "C:\Users\nzucca\ouster_example\python\build\temp.win-amd64-cpython-311\Release\CMakeFiles\3.21.1\VCTargetsPath.vcxproj" on node 1 (default targets).
          C:\Program Files (x86)\Microsoft Visual Studio\2017\BuildTools\Common7\IDE\VC\VCTargets\Microsoft.Cpp.WindowsSDK.targets(46,5): error MSB8036: The Windows SDK version 8.1 was not found. Install the required version of Windows SDK or change the SDK version in the project property pages or by right-clicking the solution and selecting "Retarget solution". [C:\Users\nzucca\ouster_example\python\build\temp.win-amd64-cpython-311\Release\CMakeFiles\3.21.1\VCTargetsPath.vcxproj]
          Done Building Project "C:\Users\nzucca\ouster_example\python\build\temp.win-amd64-cpython-311\Release\CMakeFiles\3.21.1\VCTargetsPath.vcxproj" (default targets) -- FAILED.

          Build FAILED.

          "C:\Users\nzucca\ouster_example\python\build\temp.win-amd64-cpython-311\Release\CMakeFiles\3.21.1\VCTargetsPath.vcxproj" (default target) (1) ->
          (_CheckWindowsSDKInstalled target) ->
            C:\Program Files (x86)\Microsoft Visual Studio\2017\BuildTools\Common7\IDE\VC\VCTargets\Microsoft.Cpp.WindowsSDK.targets(46,5): error MSB8036: The Windows SDK version 8.1 was not found. Install the required version of Windows SDK or change the SDK version in the project property pages or by right-clicking the solution and selecting "Retarget solution". [C:\Users\nzucca\ouster_example\python\build\temp.win-amd64-cpython-311\Release\CMakeFiles\3.21.1\VCTargetsPath.vcxproj]

              0 Warning(s)
              1 Error(s)

          Time Elapsed 00:00:00.64

        Exit code: 1

[... 35 lines cut ...]
        File "C:\Users\nzucca\AppData\Local\Temp\pip-build-env-75xmxof4\overlay\Lib\site-packages\setuptools\_distutils\cmd.py", line 318, in run_command
          self.distribution.run_command(command)
        File "C:\Users\nzucca\AppData\Local\Temp\pip-build-env-75xmxof4\overlay\Lib\site-packages\setuptools\dist.py", line 1244, in run_command
          super().run_command(command)
        File "C:\Users\nzucca\AppData\Local\Temp\pip-build-env-75xmxof4\overlay\Lib\site-packages\setuptools\_distutils\dist.py", line 988, in run_command
          cmd_obj.run()
        File "C:\Users\nzucca\AppData\Local\Temp\pip-build-env-75xmxof4\overlay\Lib\site-packages\setuptools\_distutils\command\build.py", line 131, in run
          self.run_command(cmd_name)
        File "C:\Users\nzucca\AppData\Local\Temp\pip-build-env-75xmxof4\overlay\Lib\site-packages\setuptools\_distutils\cmd.py", line 318, in run_command
          self.distribution.run_command(command)
        File "C:\Users\nzucca\AppData\Local\Temp\pip-build-env-75xmxof4\overlay\Lib\site-packages\setuptools\dist.py", line 1244, in run_command
          super().run_command(command)
        File "C:\Users\nzucca\AppData\Local\Temp\pip-build-env-75xmxof4\overlay\Lib\site-packages\setuptools\_distutils\dist.py", line 988, in run_command
          cmd_obj.run()
        File "<string>", line 47, in run
        File "<string>", line 84, in build_extension
        File "C:\Program Files\WindowsApps\PythonSoftwareFoundation.Python.3.11_3.11.1008.0_x64__qbz5n2kfra8p0\Lib\subprocess.py", line 413, in check_call
          raise CalledProcessError(retcode, cmd)
      subprocess.CalledProcessError: Command '['cmake', 'C:\\Users\\nzucca\\ouster_example\\python', '-DCMAKE_LIBRARY_OUTPUT_DIRECTORY=C:\\Users\\nzucca\\ouster_example\\python\\build\\lib.win-amd64-cpython-311\\ouster\\', '-DPYTHON_EXECUTABLE=C:\\Users\\nzucca\\AppData\\Local\\Microsoft\\WindowsApps\\PythonSoftwareFoundation.Python.3.11_qbz5n2kfra8p0\\python.exe', '-DCMAKE_BUILD_TYPE=Release', '-DOUSTER_SDK_PATH=C:\\Users\\nzucca\\ouster_example\\']' returned non-zero exit status 1.
      [end of output]

  note: This error originates from a subprocess, and is likely not a problem with pip.
  ERROR: Failed building wheel for ouster-sdk
Failed to build ouster-sdk
ERROR: Failed to build one or more wheels`

Nico-Trimble, 2023-06-16:
Solved: I was using python 3.11, which I noticed some people having problems with. I downgraded to python 3.9 and everything worked. 

Might I suggest adding a note to [https://static.ouster.dev/sdk-docs/python/devel.html](https://static.ouster.dev/sdk-docs/python/devel.html) indicating that python3.11 isn't supported, or might have problems? Such a note would have saved me a ton of time.

I won't close this in case someone thinks this qualifies as a bug or issue that needs to be resolved, but as far as I'm concerned the problem is solved.

kairenw, 2023-06-16:
Hi @Nico-Trimble ,

Sorry about that. I've updated your ticket. We hope to release Python 3.11 wheels within the next 3-4 weeks. I will try to get around the updating the docs before then to specify that there are no Python 3.11 wheels available for now.

kairenw, 2023-07-09:
Python 3.11 wheels are now available via Pypi - for verison 0.9.0

closed 2023-07-09 by kairenw