ouster-lidar/ouster-sdk #528: Is it possible to get the Netmask of a device?

Nico-Trimble, 2023-06-19:
**Describe your question**
I've been able to set the static IP and net mask of my ouster lidars using the instructions provided [here](https://forum.ouster.at/d/63-how-i-can-assign-static-ip-to-os1), but I haven't been able to retrieve the configured net mask. Is this possible in the sdk?

kairenw, 2023-06-20:
You should be able to retrieve it by using the sensor's HTTP API directly (instead of through the SDK).

for ex, for my sensor at 10.0.0.168, trying:
```
$ curl -X GET 10.0.0.168/api/v1/system/network/ipv4/override
"10.0.0.168/24"
```
I get back the static IP and the subnet mask. Hope that helps!

Nico-Trimble, 2023-06-20:
That's exactly what I'm looking for, and it worked for me. Thanks!

closed 2023-06-20 by Nico-Trimble