ouster-lidar/ouster-sdk #529: Help with No module named 'ouster.client._client'

schep87, 2023-06-26:
I followed the install guide for Windows for the Ouster SDK in python. I need help resolving the issue below.

When I try to run any of the examples, I get the following errors:
` python -m ouster.sdk.examples.pcap "C:\git\ouster\pcap_to_csv\20230619_1302_OS-0-32-U3_122228000279.pcap" "C:\git\ouster\pcap_to_csv\20230619_1302_OS-0-32-U3_122228000279.json" pcap-to-csv --scan-num 5`

Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Users\user\AppData\Local\Programs\Python\Python311\Lib\site-packages\ouster\sdk\examples\pcap.py", line 18, in <module>
    from ouster import client, pcap
  File "C:\Users\user\AppData\Local\Programs\Python\Python311\Lib\site-packages\ouster\client\__init__.py", line 9, in <module>
    from ._client import SensorInfo
ModuleNotFoundError: No module named 'ouster.client._client'

`simple-viz --pcap "C:\git\ouster\pcap_to_csv\20230619_1302_OS-0-32-U3_122228000279.pcap" --meta "C:\git\ouster\pcap_to_csv\20230619_1302_OS-0-32-U3_122228000279.json"`

Traceback (most recent call last):
  File "<frozen runpy>", line 198, in _run_module_as_main
  File "<frozen runpy>", line 88, in _run_code
  File "C:\Users\user\AppData\Local\Programs\Python\Python311\Scripts\simple-viz.exe\__main__.py", line 4, in <module>
  File "C:\Users\user\AppData\Local\Programs\Python\Python311\Lib\site-packages\ouster\sdk\simple_viz.py", line 9, in <module>
    from ouster import client
  File "C:\Users\user\AppData\Local\Programs\Python\Python311\Lib\site-packages\ouster\client\__init__.py", line 9, in <module>
    from ._client import SensorInfo
ModuleNotFoundError: No module named 'ouster.client._client'

**Platform**
- Programming Language - Python
  - Version 3.11.1
- Operating System - Windows 10
- Machine Architecture - x64

kairenw, 2023-06-27:
hi @schep87 are you sure your install worked? we don't currently have python 3.11 wheels? or you built it yourself?

schep87, 2023-06-27:
Hi @kairenw,

I followed the instructions and had to do several iterations of choosing the correct packages within Visual Studio Installer and vcpkg. I was then able to get it to install using `pip install ouster-sdk`

```
c:\git>python --version
Python 3.11.1

c:\git>pip list | grep ouster
ouster-sdk                   0.8.1
```

If Ouster-SDK is not supported by Python 3.11, then PLEASE FIX YOUR DOCUMENTATION. According to this page, 3.11 should be supported: https://static.ouster.dev/sdk-docs/installation.html

[image]

kairenw, 2023-06-27:
Apologies. Should be updated now

schep87, 2023-06-27:
Thanks for updating that. I will close this now, as it is probably directly related to using Python version 3.11.