ouster-lidar/ouster-sdk #535: Installed SDK linking error against libpcap

akloeker, 2023-07-11:
I have installed the Ouster SDK inside an Ubuntu 22.04 Docker container using this script:
```bash
#!/bin/bash

# Install Ouster SDK
OUSTER_SDK_REPO="https://github.com/ouster-lidar/ouster_example.git"
OUSTER_SDK_VERSION="20230710"
OUSTER_SDK_TEMP_DIR="/ouster_sdk"
OUSTER_SDK_INSTALL_DIR="/usr/local"

git clone --branch $OUSTER_SDK_VERSION $OUSTER_SDK_REPO $OUSTER_SDK_TEMP_DIR
cd $OUSTER_SDK_TEMP_DIR
cmake -DCMAKE_INSTALL_PREFIX=$OUSTER_SDK_INSTALL_DIR -DBUILD_PCAP=ON -DBUILD_OSF=OFF -DBUILD_VIZ=OFF -DBUILD_EXAMPLES=OFF -DCMAKE_BUILD_TYPE=Release .
cmake --build . --target install
rm -r $OUSTER_SDK_TEMP_DIR
echo "export CMAKE_PREFIX_PATH=$OUSTER_SDK_INSTALL_DIR" >> /root/.bashrc
```
All the dependencies from your docs are installed with apt. Installing the Ouster SDK works flawlessly. However, when I want to use it in another project I link it in my CMakeLists.txt as followed:
```CMake
...
find_package(OusterSDK REQUIRED)
...
target_link_libraries(my_project
  OusterSDK::ouster_client
  OusterSDK::ouster_pcap
)
```
Doing this I get either the error
```
CMake Error at CMakeLists.txt:31 (add_executable):
  Target "my_project" links to target "libpcap::libpcap" but the target was not
  found.  Perhaps a find_package() call is missing for an IMPORTED target, or
  an ALIAS target is missing?
```
or
```
/usr/bin/ld: cannot find -llibpcap::libpcap: No such file or directory
collect2: error: ld returned 1 exit status
```
`libpcap-dev` is installed and `/lib/x86_64-linux-gnu/libpcap.a` exists.
What's wrong here?

chrisbayruns, 2023-07-12:
Hello @akloeker,
In order to better diagnose your issue, can you post your:
1. Dockerfile
2. How you built/installed ouster_sdk
3. The full error log

If you would rather not have this info on the open internet, you can create a support ticket  over here https://ouster.atlassian.net/servicedesk/customer/portal/8/create/86 and post the information there. When creating the ticket, mention the github issue and my name on the ticket.

akloeker, 2023-07-12:
Hello @chrisbayruns, thanks for reaching out. Here is a minimum example to reproduce the error. I have also attached a .zip file with all necessary files and information: [ouster_sdk_test.zip](https://github.com/ouster-lidar/ouster_example/files/12027587/ouster_sdk_test.zip)

**Dockerfile with ouster_sdk installation**
```Dockerfile
FROM rwthika/ros2

RUN apt-get update && apt-get install -y \
    libeigen3-dev \
    libjsoncpp-dev \
    libtins-dev \
    libpcap-dev \
    libcurl4-openssl-dev

ARG OUSTER_SDK_REPO="https://github.com/ouster-lidar/ouster_example.git"
ARG OUSTER_SDK_VERSION="20230710"
ARG OUSTER_SDK_TEMP_DIR="/ouster_sdk"
ARG OUSTER_SDK_INSTALL_DIR="/usr/local"

RUN git clone --branch $OUSTER_SDK_VERSION $OUSTER_SDK_REPO $OUSTER_SDK_TEMP_DIR
RUN cd $OUSTER_SDK_TEMP_DIR && \
    cmake -DCMAKE_INSTALL_PREFIX=$OUSTER_SDK_INSTALL_DIR -DBUILD_PCAP=ON -DBUILD_OSF=OFF -DBUILD_VIZ=OFF -DBUILD_EXAMPLES=OFF -DCMAKE_BUILD_TYPE=Release . && \
    cmake --build . --target install
RUN rm -r $OUSTER_SDK_TEMP_DIR
RUN echo "export CMAKE_PREFIX_PATH=$OUSTER_SDK_INSTALL_DIR" >> /root/.bashrc
```

**CMakeLists.txt**
```CMake
cmake_minimum_required(VERSION 3.10)
project(MyProject)

find_package(OusterSDK REQUIRED)

add_executable(MyProject main.cpp)

target_link_libraries(MyProject
    OusterSDK::ouster_client
    OusterSDK::ouster_pcap
)
```

**main.cpp**
```cpp
// Ouster SDK
#include <ouster/client.h>
#include <ouster/lidar_scan.h>
#include <ouster/os_pcap.h>

int main(int argc, char** argv)
{
    // not necessary for testing
}
```

**Steps that were executed**
```bash
docker build -t ouster_sdk_test -f Dockerfile .
docker run --rm -it -v ./:/docker-ros/ws ouster_sdk_test
# Inside the Docker container
cmake .
```

**Full error log**
```
root@6f58cccd4ac3:/docker-ros/ws# cmake .
-- The C compiler identification is GNU 11.3.0
-- The CXX compiler identification is GNU 11.3.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /usr/bin/cc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /usr/bin/c++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Found OusterSDK: /usr/local/lib/cmake/OusterSDK/OusterSDKConfig.cmake
-- Found CURL: /usr/lib/x86_64-linux-gnu/libcurl.so (found version "7.81.0")  
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD
-- Performing Test CMAKE_HAVE_LIBC_PTHREAD - Success
-- Found Threads: TRUE  
-- Configuring done
CMake Error at CMakeLists.txt:6 (add_executable):
  Target "MyProject" links to target "libpcap::libpcap" but the target was
  not found.  Perhaps a find_package() call is missing for an IMPORTED
  target, or an ALIAS target is missing?

-- Generating done
CMake Generate step failed.  Build files cannot be regenerated correctly.
```

kairenw, 2023-07-13:
Hi @akloeker 

Thanks for providing the files. We believe it's a change we made with the latest release - so go ahead and use the last tagged release (there should be no difference in supported FW)

We'll get back to you next week (towards the end of the week) as we look into this.

akloeker, 2023-07-14:
Thaks, I can confirm that version `20230403` works fine.

chrisbayruns, 2023-07-20:
Hello @akloeker,
I have put together an example that should work with your intended build flow until we can fix it in a following release.
[libpcap_temporary_fix.zip](https://github.com/ouster-lidar/ouster_example/files/12114443/libpcap_temporary_fix.zip)

akloeker, 2023-11-08:
Version `20231031` works just fine, and fixes the problem 👍

closed 2023-11-08 by akloeker