ouster-lidar/ouster-sdk #540: Unable to get an intensity image that shows the environment similar to Ouster Studio intensity image using C++ code

arunumd, 2023-07-25:
**Describe your question**

The Ouster Studio has the feature of viewing the 2D intensity image on the screen. This image looks very realistic. It shows the environment of the LiDAR like a photograph. So I am trying to reproduce the same kind of image through some C++ and OpenCV code by following the docs, provided examples and inventing some more things of my own. The code below is what I have produced for performing this operation.

```#include <chrono>
#include <fstream>
#include <iomanip>
#include <iostream>
#include <memory>
#include <string_view>
#include <vector>
#include <opencv2/core.hpp>
#include <opencv2/highgui.hpp>

#include "ouster/client.h"
#include "ouster/impl/build.h"
#include "ouster/lidar_scan.h"
#include "ouster/types.h"

using namespace ouster;

const size_t UDP_BUF_SIZE = 65536;

void FATAL(const char* msg) {
    std::cerr << msg << std::endl;
    std::exit(EXIT_FAILURE);
}

//! [docs-stag-x-image-form]
img_t<double> get_x_in_image_form(const LidarScan& scan, bool destaggered,
                                  const sensor::sensor_info& info) {
    // For convenience, save w and h to variables
    const size_t w = info.format.columns_per_frame;
    const size_t h = info.format.pixels_per_column;

    // Get the XYZ in ouster::Points (n x 3 Eigen array) form
    XYZLut lut = make_xyz_lut(info);
    auto cloud = cartesian(scan.field(sensor::ChanField::RANGE), lut);

    // Access x and reshape as needed
    // Note that the values in cloud.col(0) are ordered
    auto x = Eigen::Map<const img_t<double>>(cloud.col(0).data(), h, w);
    auto x_destaggered = destagger<double>(x, info.format.pixel_shift_by_row);

    // Apply destagger if desired
    if (!destaggered) return x;
    return x_destaggered;
}
//! [docs-etag-x-image-form]
//

int main(int argc, char* argv[]) {
    std::cerr
        << "Version: " << ouster::SDK_VERSION_FULL << " ("
        << ouster::BUILD_SYSTEM << ")"
        << "\n\nUsage: client_example <sensor_hostname> [<udp_destination>]"
           "\n\n<udp_destination> is optional: leave blank for "
           "automatic destination detection"
        << std::endl;
    // Limit ouster_client log statements to "info" and direct the output to log
    // file rather than the console (default).
    sensor::init_logger("info", "ouster.log");
    const std::string sensor_hostname =
        "169.254.19.20"; //"169.254.181.60";  //"os-122124000462.local";
    const std::string data_destination = "";
    std::cerr << "Connecting to \"" << sensor_hostname << "\"...\n";
    auto handle = sensor::init_client(sensor_hostname, data_destination);
    if (!handle) FATAL("Failed to connect");
    std::cerr << "Connection to sensor succeeded" << std::endl;

    std::cerr << "Gathering metadata..." << std::endl;
    auto metadata = sensor::get_metadata(*handle);
    auto os1_client_version = ouster::sensor::client_version();
    std::cout << os1_client_version << std::endl;

    // Raw metadata can be parsed into a `sensor_info` struct
    sensor::sensor_info info = sensor::parse_metadata(metadata);

    size_t w = info.format.columns_per_frame;
    size_t h = info.format.pixels_per_column;

    ouster::sensor::ColumnWindow column_window = info.format.column_window;

    std::cerr << "  Firmware version:  " << info.fw_rev
              << "\n  Serial number:     " << info.sn
              << "\n  Product line:      " << info.prod_line
              << "\n  Scan dimensions:   " << w << " x " << h
              << "\n  Column window:     [" << column_window.first << ", "
              << column_window.second << "]" << std::endl;

    LidarScan scan = LidarScan{w, h, info.format.udp_profile_lidar};

    // A ScanBatcher can be used to batch packets into scans
    sensor::packet_format pf = sensor::get_format(info);
    ScanBatcher batch_to_scan(info.format.columns_per_frame, pf);

    /*
     * The network client provides some convenience wrappers around socket APIs
     * to facilitate reading lidar and IMU data from the network. It is also
     * possible to configure the sensor offline and read data directly from a
     * UDP socket.
     */
    std::cerr << "Capturing points... ";

    // buffer to store raw packet data
    auto packet_buf = std::make_unique<uint8_t[]>(UDP_BUF_SIZE);

    sensor::client_state st = sensor::poll_client(*handle);

    // check for error status
    if (st & sensor::CLIENT_ERROR) FATAL("Sensor client returned error state!");

    // check for lidar data, read a packet and add it to the current batch
    if (st & sensor::LIDAR_DATA) {
        if (!sensor::read_lidar_packet(*handle, packet_buf.get(), pf)) {
            FATAL("Failed to read a packet of the expected size!");
        }

        // batcher will return "true" when the current scan is complete
        if (batch_to_scan(packet_buf.get(), scan)) {
            // retry until we receive a full set of valid measurements
            // (accounting for azimuth_window settings if any)
            if (scan.complete(info.format.column_window)) {};
        }
    }

    // check if IMU data is available (but don't do anything with it)
    if (st & sensor::IMU_DATA) {
        sensor::read_imu_packet(*handle, packet_buf.get(), pf);
    }

    std::cerr << "ok" << std::endl;

    XYZLut lut = make_xyz_lut(info);

    auto range = scan.field(sensor::ChanField::RANGE);
    auto cloud = cartesian(range, lut);

    // 3. Destaggering
    // Fields come in w x h arrays, but they are staggered, so that a column
    // reflects the timestamp. To get each column to make visual sense,
    // destagger the image
    std::cerr
        << "\n3. Getting staggered and destaggered images of Reflectivity..."
        << std::endl;

    Eigen::Array<uint32_t, -1, -1, Eigen::RowMajor> reflectivity;

    if (info.format.udp_profile_lidar ==
        sensor::UDPProfileLidar::PROFILE_LIDAR_LEGACY) {
        reflectivity = scan.field(sensor::ChanField::REFLECTIVITY);
    } else if (info.format.udp_profile_lidar ==
               sensor::UDPProfileLidar::PROFILE_RNG19_RFL8_SIG16_NIR16_DUAL) {
        reflectivity = scan.field<uint8_t>(sensor::ChanField::REFLECTIVITY)
                           .cast<uint32_t>();
    } else {  // legacy or single return profile
        reflectivity = scan.field<uint16_t>(sensor::ChanField::REFLECTIVITY)
                           .cast<uint32_t>();
    }

    //! [doc-stag-cpp-destagger]
    auto reflectivity_destaggered =
        destagger<uint32_t>(reflectivity, info.format.pixel_shift_by_row);
    //! [doc-etag-cpp-destagger]

    auto x_image_staggered = get_x_in_image_form(scan, false, info);
    auto x_image_destaggered = get_x_in_image_form(scan, true, info);
    cv::Mat intensity_img = cv::Mat(h, w, CV_8UC3);

    for (size_t i = 0; i < h; i++) {
        for (size_t j = 0; j < w; j++) {
            const auto print_row = i;
            const auto print_column = j;
            intensity_img.at<char>(print_row, print_column) = reflectivity_destaggered(print_row, print_column);
        }
        std::cout << "\n";
    }
    cv::namedWindow("Intensity image", cv::WINDOW_AUTOSIZE);
    cv::imshow("Intensity image", intensity_img);
    cv::waitKey(0);
    return 0;
}
```

However the image I get is nothing like what I get from Ouster Studio.  Please see the snapshots below for comparison.

**Screenshots**

**What I get from Ouster Studio intensity image:**
[image]

**What I get from my C++ code using OpenCV and Ouster SDK:**
[image]

It seems as though something is wrong with this code in terms of how I try to get the reflectivity array for the environment. I am wondering what is wrong?

**Platform (please complete the following information):**
- Ouster Sensor? \[OS-1 128 channel Rev-7]
- Ouster Firmware Version? \[3.0.1]
- Programming Language? \[C++]
- Operating System? \[Linux and WSL using XServer]
- Machine Architecture? \[x64]
- git commit hash (if not the latest).

arunumd, 2023-07-26:
Hello,

Any idea why does the reflectivity matrix have mostly zeros in its entries? When I `std::cout` the `intensity_img`, it looks like this. Mostly overwhelmed by zeroes. So the image looks mostly black.

[image]

Samahu, 2023-07-26:
There might be different issues with your code:
- First I think you should start by applying auto leveling to each channel before you try to pass the values to an RGB image with uint8 per channel.
  - You can find information on **auto leveling** and how to apply in this [Google Colab Notebook](https://github.com/ouster-lidar/ouster-yolov5-demo/blob/main/Ouster_Yolo5_Demo.ipynb). 
- Second if the range of values for the reflectivity channel goes beyond 255 so you may need to re-scale or re-map the values into the range of [0, 255] before you copy the values into a uint8 otherwise the value will be truncated. You can check that by searching for the maximum value in the reflectivity image.
- Lastly based on the pattern of pixels shown in the resulting images it appears there are some problem with the order of how you copy the pixels. The shared [Google Colab Notebook](https://github.com/ouster-lidar/ouster-yolov5-demo/blob/main/Ouster_Yolo5_Demo.ipynb) although it is written in python but can provide with guidance on how to a channel to an OpenCV **Image** or **cv::Mat** object.

Hope this helps!

Samahu, 2023-07-31:
No response, marking the issue as done.

closed 2023-07-31 by Samahu