ouster-lidar/ouster-sdk #549: Read Access Violation in ouster::sensor::parse_config

Misamag, 2023-09-13:
**Description**

When calling 

```
ouster::sensor::sensor_config config;
if (!ouster::sensor::get_config(hostname, config))
...
```

from my application, I get a read access violation in 

```
static sensor_config parse_config(const Json::Value& root) {
    sensor_config config{};

    if (!root["udp_dest"].empty()) {
        config.udp_dest = root["udp_dest"].asString();
```

The config read in client.cpp:285 `auto res = sensor_http->get_config_params(active)` is a valid JSON string.

It looks like the parsing in types.cpp:659 `if (!Json::parseFromStream(builder, ss, &root, &errors)) {` results in an invalid root. The `ss` argument contains the JSON string but root or root["udp_dest"], after this call, just contain garbage.

**Platform**
- OS2
- Ouster Firmware Version? 2.5.2
- Programming Language? C++
- Operating System? Windows 10 22H2
- Machine Architecture? x64
- git commit hash 9d0971107f6f9c95e16afd727fa2534d01a0fe4e

** Note **
* I upgraded from an older version of the API and followed up the docs here: https://static.ouster.dev/sdk-docs/cpp/building.html
* Application using API is compiled for c++20 using VS build tools for 2019 from within VS 2022. 
* Using Qt libs 5.4 as well.

twslankard, 2023-09-13:
Thanks for the detailed report, @Misamag.

There may be an issue with how ouster-sdk is interacting with JsonCpp here.

If possible, could you provide 
1. the contents of the JSON string you obtained when this issue occurred,
2. the version of JsonCpp you're using,
3. whether you're building a Debug build or Release build

Best Regards,

Tom
Staff Engineer
Ouster Inc.

Misamag, 2023-09-14:
@twslankard,

this is the JSON string: 
```
{"phase_lock_offset": 0, "udp_profile_lidar": "LEGACY", "nmea_baud_rate": "BAUD_9600", "udp_port_imu": 53510, "udp_dest": "239.83.104.101", "sync_pulse_out_angle": 360, "nmea_leap_seconds": 0, "sync_pulse_out_polarity": "ACTIVE_HIGH", "udp_port_lidar": 53509, "lidar_mode": "2048x10", "timestamp_mode": "TIME_FROM_INTERNAL_OSC", "sync_pulse_out_pulse_width": 0, "signal_multiplier": 3, "operating_mode": "NORMAL", "sync_pulse_out_frequency": 1, "sync_pulse_in_polarity": "ACTIVE_HIGH", "phase_lock_enable": false, "multipurpose_io_mode": "OFF", "nmea_in_polarity": "ACTIVE_HIGH", "azimuth_window": [300000, 50000], "udp_profile_imu": "LEGACY", "nmea_ignore_valid_char": 0, "columns_per_packet": 16}
```

JsonCpp: 1.9.5

Issue persiststs in Debug build. In Release build, it parses correctly.

Thanks,
Marvin

Misamag, 2023-09-15:
I have just noticed something: when I switch the language version in the cmake file from 14 to 17, I do not get any errors. The following is just a guess as I have not made much of reverse-engineering:

* My calling application (which is C++17) is using the sensor_config and interprets the optional<T> values as std::optional
* The lib, as compiled with C++14 support uses nonstd::optional during build
* no binary compatibility between those types => corruption

Samahu, 2023-09-18:
@Misamag this is the same issue as #483 and yes it happens when the nonstd::optional is compiled with different cxx standards. The fix to this problem is within the application embedding ouster client.

closed 2023-09-18 by Samahu

Misamag, 2023-09-19:
@Samahu thanks, I see. It would be handy to have such information available in the starting guide at https://static.ouster.dev/sdk-docs/cpp/building.html. Like the required C++ version, mandatory defines (like EIGEN_MAX_ALIGN_BYTES=32) an so on.