ouster-lidar/ouster-sdk #550: Unable to convert pcap to pcd/bin files with existing command from "https://static.ouster.dev/sdk-docs/python/examples/conversion.html"

jyothsna-phd22, 2023-09-14:
I have collected the data using Ouster OS-1 32-channel LIDAR. I am trying to convert it to a pcd file. Earlier, I was able to convert my data to pcd/ply/las/csv of all the frames using the link "https://static.ouster.dev/sdk-docs/python/examples/conversion.html". But now, some errors are being observed.

Kindly help me resolve the issue. 

[image]

twslankard, 2023-09-14:
Thanks for the report, @jyothsna-phd22.

Can you tell me whether you have the `open3d` package installed? It looks like you're using `conda` so try running `conda list` to see if `open3d` is listed. If it isn't, you could try installing it and then re-running the command.

Also, if the command worked before but doesn't now, perhaps there is another python environment on the system you were using that already has `open3d` installed.

I hope this has been helpful. Please let me know if you continue to experience issues, and if so, please include as much of the info below as possible.

**Platform (please complete the following information):**
- Ouster Sensor? \[e.g. OS-0, OS-1, ..\]
- Ouster Firmware Version? \[e.g. 2.3, 2.4, ..\]
- Programming Language? \[e.g. C++, Python, ..\]
  - For Python include the targeted Python version
- Operating System? \[e.g. Linux, Windows, MacOS\]
- Machine Architecture? \[e.g. x64, arm\]
- git commit hash (if not the latest).

Best Regards,

Tom
Staff Engineer
Ouster Inc

jyothsna-phd22, 2023-09-16:
I have open3d installed in my anaconda prompt.
[image]

Platform (please complete the following information):

Ouster Sensor [OS-1 32 channel]
Ouster Firmware Version? [v2.4.0]
Programming Language? [ Python]
Python Version: Python 3.9.13
Operating System? [Windows]
Machine Architecture? [x64]

jyothsna-phd22, 2023-09-16:
It's working, Thank You. I have another question.  Can you please suggest to me the data labeling tool for streaming data? And is there any work done on ouster lidar on real-time 3D object detection?