ouster-lidar/ouster-sdk #551: Confusing parameter documentation in mtp_init_client

Misamag, 2023-09-14:
In function mtp_init_client, there is a `mtp_dest_host` parameter. The doc is `multicast ip address where the sensor should send data`.

As I can see from the code, it's rather the local ip address of the interface which is used to join the mutlicast group which is defined in the config parameter. If it's empty, the OS will automatically all (?) interfaces.

twslankard, 2023-09-14:
Thanks for the report @Misamag - I'll make sure we fix the documentation.

Best Regards,

Tom
Staff Engineer
Ouster Inc.

twslankard, 2023-09-14:
After double-checking the code, it looks like `mtp_dest_host` specifies the address of the interface used to join the multicast group.

However, any interface will be used[1] if `mtp_dest_host` is empty (which as you've probably seen in the code sets the  address to `INADDR_ANY` in an `ip_mreq` structure passed to `setsockopt` to join the multicast group.)

[1] https://man7.org/linux/man-pages/man7/ip.7.html