ouster-lidar/ouster-sdk #559: Odometry data from Pcap files

BIGINCHI, 2023-10-12:
Is there any way to obtain Odometry data to know if my lidar pcap file is moving left or right, because when im viewing the frames all the frames update at the 0,0,0 origin and has no translation. so i want to know if there is any method to extract translation and rotation data from the ouster files.

[image]

so instead of all files merging at the center i want to have all the frames updated, overlapped and stitched one after another, i have tried using icp method to combine all the files but it has been ineffective and very slow with in accurate 3d reconstructions. Any help regarding this would be of great help.

**Platform (please complete the following information):**
- Ouster Sensor?  OS-0
- Ouster Firmware Version? 2.4
- Programming Language? Python
  - python  3.9
- Operating System? Windows
- Machine Architecture? x64

bexcite, 2023-12-06:
Hi @BIGINCHI,

You can try kiss-icp integration to get poses and then use `--kitti-poses` param for `ouster-cli viz` to visualize registered 3d point cloud:

```
pip install ouster-sdk kiss-icp
kiss_icp_pipeline --deskew ./OS-0-128_v3.0.1_1024x10.pcap
ouster-cli source ./OS-0-128_v3.0.1_1024x10.pcap viz --kitti-poses ./results/latest/OS-0-128_v3.0.1_poses_kitti.txt --accum-map
```

[image]

It's all the same viz: https://static.ouster.dev/sdk-docs/python/viz/viz-scans-accum.html#viz-scans-accum

twslankard, 2024-07-23:
This issue appears resolved. Please open another issue if you run into any further questions or issues, @BIGINCHI . Thanks!

Cheers,

Tom
Staff Engineer
Ouster Inc.

closed 2024-07-23 by twslankard