ouster-lidar/ouster-sdk #560: ouster-sdk python packages raise InvalidVersion error with upcoming version of pip

kgreenek, 2023-10-13:
PEP 440 defines a standard for the versions used in pip packages: https://peps.python.org/pep-0440/

This has been a standard since 2013, but not enforced due to compatibility concerns.

However, recently they have decided to start enforcing this standard. New versions of pip raise an InvalidVersion exception when trying to install ouster-sdk packages from pip, similar to the following:

```
(Traceback (most recent call last):
  File "/home/greg/.cache/bazel/_bazel_greg/b0c20c8ca453cbc6e375a5451bd4af5d/external/rules_ros2_python_x86_64-unknown-linux-gnu/lib/python3.10/runpy.py", line 196, in _run_module_as_main
    return _run_code(code, main_globals, None,
  File "/home/greg/.cache/bazel/_bazel_greg/b0c20c8ca453cbc6e375a5451bd4af5d/external/rules_ros2_python_x86_64-unknown-linux-gnu/lib/python3.10/runpy.py", line 86, in _run_code
    exec(code, run_globals)
  File "/home/greg/.cache/bazel/_bazel_greg/b0c20c8ca453cbc6e375a5451bd4af5d/external/rules_python/python/pip_install/tools/wheel_installer/wheel_installer.py", line 200, in <module>
    main()
  File "/home/greg/.cache/bazel/_bazel_greg/b0c20c8ca453cbc6e375a5451bd4af5d/external/rules_python/python/pip_install/tools/wheel_installer/wheel_installer.py", line 192, in main
    _extract_wheel(
  File "/home/greg/.cache/bazel/_bazel_greg/b0c20c8ca453cbc6e375a5451bd4af5d/external/rules_python/python/pip_install/tools/wheel_installer/wheel_installer.py", line 132, in _extract_wheel
    whl_deps = sorted(whl.dependencies(extras_requested) - self_edge_dep)
  File "/home/greg/.cache/bazel/_bazel_greg/b0c20c8ca453cbc6e375a5451bd4af5d/external/rules_python/python/pip_install/tools/wheel_installer/wheel.py", line 79, in dependencies
    if req.marker is None or any(
  File "/home/greg/.cache/bazel/_bazel_greg/b0c20c8ca453cbc6e375a5451bd4af5d/external/rules_python/python/pip_install/tools/wheel_installer/wheel.py", line 80, in <genexpr>
    req.marker.evaluate({"extra": extra})
  File "/home/greg/.cache/bazel/_bazel_greg/b0c20c8ca453cbc6e375a5451bd4af5d/external/pypi__setuptools/pkg_resources/_vendor/packaging/markers.py", line 252, in evaluate
    return _evaluate_markers(self._markers, current_environment)
  File "/home/greg/.cache/bazel/_bazel_greg/b0c20c8ca453cbc6e375a5451bd4af5d/external/pypi__setuptools/pkg_resources/_vendor/packaging/markers.py", line 158, in _evaluate_markers
    groups[-1].append(_eval_op(lhs_value, op, rhs_value))
  File "/home/greg/.cache/bazel/_bazel_greg/b0c20c8ca453cbc6e375a5451bd4af5d/external/pypi__setuptools/pkg_resources/_vendor/packaging/markers.py", line 116, in _eval_op
    return spec.contains(lhs, prereleases=True)
  File "/home/greg/.cache/bazel/_bazel_greg/b0c20c8ca453cbc6e375a5451bd4af5d/external/pypi__setuptools/pkg_resources/_vendor/packaging/specifiers.py", line 568, in contains
    normalized_item = _coerce_version(item)
  File "/home/greg/.cache/bazel/_bazel_greg/b0c20c8ca453cbc6e375a5451bd4af5d/external/pypi__setuptools/pkg_resources/_vendor/packaging/specifiers.py", line 36, in _coerce_version
    version = Version(version)
  File "/home/greg/.cache/bazel/_bazel_greg/b0c20c8ca453cbc6e375a5451bd4af5d/external/pypi__setuptools/pkg_resources/_vendor/packaging/version.py", line 198, in __init__
    raise InvalidVersion(f"Invalid version: '{version}'")
pkg_resources.extern.packaging.version.InvalidVersion: Invalid version: '#34~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Thu Sep  7 13:12:03 UTC 2'
) error code: '1'
```

More details about the decision from the pip maintainers can be found here:
https://github.com/pypa/pip/issues/12012

twslankard, 2023-10-13:
Thanks for the report, @kgreenek .

I'm not able to reproduce this issue with the latest production version of `pip` (23.2.1.)

At your convenience would you please provide some information about which versions of `pip` and `python3` you are using?

Best Regards,
Tom

Staff Engineer
Ouster Inc.

kgreenek, 2023-10-13:
I am using bazel with [rules_python](https://github.com/bazelbuild/rules_python), which is admittedly an unusual setup. Using rules_python, the version exception is raised starting with version 0.26.0. Falling back to 0.25.0 continues to work.

According to [this issue](https://github.com/pypa/pip/issues/12012) on the pip github repo: "In 23.2, pip is starting to deprecate recognision of non-standard version strings. [...] The deprecated parsing logic is expected to be entirely removed in 24.0"

twslankard, 2023-10-13:
@kgreenek thanks for the additional information. Do you have a minimum reproducible example I can try?

kgreenek, 2023-10-13:
Here is a minimal reproducible example with instructions for how to run: https://github.com/kgreenek/ouster_sdk_pep440

kgreenek, 2023-10-13:
The issue appears to be that this version (`#34~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Thu Sep  7 13:12:03 UTC 2`) for ouster-sdk is not compatible with PEP 440.

I am not sure where that version string is coming from to be honest. If that version string is part of the ouster-sdk package in pypi, I would ask if you would be willing to make a change to be in compliance with PEP 440. Otherwise, I need to dig in a bit deeper and figure out where that string is coming from.

twslankard, 2023-10-13:
@kgreenek `#34~22.04.1-Ubuntu SMP PREEMPT_DYNAMIC Thu Sep  7 13:12:03 UTC 2` is not an `ouster-sdk` version. Our current `ouster-sdk` release is `0.9.0`.

kgreenek, 2023-10-13:
Interesting. It seems like there is some interaction between rules_python and the ouster-sdk wheel going on. I'll file an issue with the rules_python folks too and see if they have any insight to add.

twslankard, 2023-10-13:
@kgreenek I've narrowed this down to the dependency spec for `scipy` that is present in ouster-sdk's `setup.py`.

The current spec is this, which I believe to be a valid dependency spec per [PEP 508](https://peps.python.org/pep-0508/).

```
scipy >=1.7, <2;platform_system != "Darwin" or platform_machine != "arm64" or platform_version >= "21.0.0"
```

This dependency spec is used because there is no `scipy` available for ARM64 macOS 11 systems (though macOS 11 is technically supported by ouster-sdk.)

Although I haven't confirmed this to be the case, I suspect that there is a bug in the toolchain you're using, since it is dependent, ostensibly, on the the `platform_version` values provided by Linux conforming to PEP 440.

A workaround would be to fork ouster-sdk, remove the `scipy` dependency spec, and build a new wheel for your app. Alternatively, you could follow up to confirm whether there is a bug in the toolchain.

groodt, 2023-10-15:
Thanks for looking into this. I believe that this is actually an invalid specifier. See some notes here: https://github.com/bazelbuild/rules_python/issues/1491#issuecomment-1763358448

tl;dr: you probably want to be using `platform_release >= '21.0.0'` instead anyway.

lalten, 2024-01-11:
Hi, we're running into exactly the same issue. Any progress on a fix?

twslankard, 2024-01-25:
Hi @lalten,

I'm afraid we've not had the bandwidth to fix this issue yet, but we plan to address it in our next release which should be no later than end of March. I realize this is not ideal, but it is definitely a priority for us.

Best Regards,
Tom

Staff Engineer
Ouster Inc.

lalten, 2024-04-26:
Hi @twslankard, is this issue resolved in https://github.com/ouster-lidar/ouster_example/releases/tag/20240425?

twslankard, 2024-04-26:
Hi @lalten,

The issue is not fixed in this release, unfortunately. Please accept my apologies. We're working on it!

Cheers,
Tom

Staff Engineer
Ouster Inc.

twslankard, 2024-05-23:
@lalten

The team tells me our latest release should resolve this issue. I apologize for the mixed messages. Would you be willing to try https://pypi.org/project/ouster-sdk/0.11.1/ and let me know if it meets your needs?

Cheers,
Tom

twslankard, 2024-05-28:
I don't think this issue is fixable at this time, at least not without cooperation from the Bazel maintainers.

Here's what we see so far:

1. The problem is not reproducible for people using the most recent `pip` (24.0) and the most recent `setuptools` (70.0.0).
2. The `setuptools` has a method for evaluating environment markers and both implementations raise `InvalidVersion` for the marker `platform_release >= "21.0.0"'`. This method is apparently what Bazel uses.
3. I'm aware of no solutions for handling non PEP 440 `platform_release` values at install time. (Note, since `pip` install doesn't fail, `pip` must not be using the marker evaluation methods.)

twslankard, 2024-07-23:
In the latest version, we've removed the `platform_release` markers entirely. Here's hoping that Python tools will develop a fallback to string-based comparison for non PEP 440 version platform markers!

Thanks to all of you for your help investigating and for all of your patience. I'm closing this issue as resolved at this point. Please open a new issue if you encounter any new problems with the new release.

Cheers,
Tom

closed 2024-07-23 by twslankard