ouster-lidar/ouster-sdk #562: Wrong size of Reflectivity for Single Return Profile

99andreas, 2023-10-23:
Ouster SDK v0.9.0:

The size of Reflectivity for _Single Return Profile_ is 8 bits according to specification but is implemented as 16 bits in the C++ SDK (https://github.com/ouster-lidar/ouster_example/blob/master/ouster_client/src/lidar_scan.cpp#L70). This means that it is impossible to read correct Reflectivity from a lidar in Sigle Return Profile with the C++ SDK.

Since _Single Return Profile_ is becoming the new default profile when _Legacy_ is not supported anymore, this should probably have a high priority (a patch release 0.9.1?). Also I guess there is some test code missing that should have cought this bug earlier.

twslankard, 2023-10-23:
Thanks for the report @99andreas.

As far as I'm aware, there's no technical issue that prevents storing the 8-bit reflectivity values coming from the sensor's UDP packet stream into the 16-bit reflectivity channel in `LidarScan` objects. Are you seeing incorrect values when accessing reflectivity via `LidarScan`? If so, could you please provide a minimum reproducible example?

I should mention that the fact that they are differently sized fields shouldn't be a problem in and of itself: values from packets are casted to the appropriate types and assigned to scans on a per-pixel basis. Please refer to the `packet_format::block_field` and `packet_format::col_field` methods for more info about how this is done.

This said, there are known issues the low-bandwidth profile in 0.9.0 that will be addressed in the upcoming 0.10.0 release.

I hope this addresses your concerns. Please let me know if there are any questions.

Best Regards,
Tom

Staff Engineer
Ouster Inc.

99andreas, 2023-10-24:
Hi Tom and thanks for your swift response. I just want to confirm that treating it as a 16 bit works fine. BTW, it would be nice with some sort of  "known issues and workarounds" information somewhere.

twslankard, 2023-10-25:
@99andreas that's a great suggestion. Going forward we'll work toward including more information about known issues into the documentation.

Best Regards,
Tom

Staff Engineer
Ouster Inc.

twslankard, 2024-07-23:
I'm closing this issue since it appears resolved. Please open a new issue if we can be of any further assistance. Thanks!

closed 2024-07-23 by twslankard