ouster-lidar/ouster-sdk #564: Announcement: ouster sdk 0.10.0 breaking changes!

twslankard, 2023-10-30:
We're ready to release 0.10.0! Please see the PR that contains a change log here
https://github.com/ouster-lidar/ouster_example/pull/563

KamilJeziorski-TomTom, 2024-11-08:
Hello, I'm trying to adopt code which was originally written based on ouster sdk 0.9.0 for latest 0.13.0.

The code looks as follows:

```
for (int i = 0; i < sensorInfo.format.columns_per_packet; i++) {
	auto col_buf = packetFormat->nth_col(i, reinterpret_cast<const unsigned char*>(buffer));
	for (unsigned int u = 0; u < sensorInfo.format.pixels_per_column; u++) {
		auto r = packetFormat->px_range(packetFormat->nth_px(u, col_buf));
		auto reflectivity = packetFormat->px_reflectivity(packetFormat->nth_px(u, col_buf));		
		auto nir = packetFormat->px_ambient(packetFormat->nth_px(u, col_buf));
		auto signal = packetFormat->px_signal(packetFormat->nth_px(u, col_buf));
		
		<....>
	}
}
```
SDK 0.10.0 introduces braking change by removing `px_*` accessors, and I'm struggling to find a repplacement to access channes per pixel. Can you please suggest a solution?

matt-attack, 2024-11-09:
Hi,

There's really no direct replacement. The APIs have largely moved to using the ScanBatcher and using LidarScans instead to access that data rather than accessing it packet by packet.

If you are working with a specific packet format, it would be quite straightforward to decode that pixel manually given the pointer to it from `packetFormat->nth_px(u, col_buf)`. For instance a pixel in the RNG19_RFL8_SIG16_NIR16 format is represented by this struct:
```cpp
#pragma pack(1)
struct Pixel {
  uint32_t range;
  uint8_t reflectivity;
  uint8_t padding;
  uint16_t signal;
  uint16_t nir;// aka ambient
  uint16_t padding2;
};
```