ouster-lidar/ouster-sdk #581: KISS-ICP backwards compatibility

luccosta, 2024-02-28:
**Bug description**
Using pip to install `ouster-sdk` install a version of KISS-ICP that has backwards compatibility problems.

```
1 validation error for KISSConfig config_file
```

Checking the version:

```bash
> kiss_icp_pipeline --version
KISS-ICP Version: 0.3.0
```

Using KISS-ICP Version 0.2.10 it worked for me.

**To Reproduce**
Steps to reproduce the behavior (steps below are just an example):
1. Install with pip
2. Run SLAM on a PCAP file

rvenky125, 2024-03-12:
How can I install specific kiss-icp version

twslankard, 2024-03-12:
@luccosta @rvenky125 

The kiss-icp project changed their config API somewhat. We'll be addressing this in our next release (target date is end of March.)

In the meantime, see if you can fix the problem by installing the compatible version of kiss-icp:

```
pip install --force-reinstall -v "kiss-icp==0.2.10"
```

Regards,
Tom

Staff Engineer
Ouster Inc.

twslankard, 2024-07-23:
Our most recent releases (0.11.0, 0.11.1, 0.12.0) use kiss-icp 0.4.0 (the latest at this time.) Please let us know if any future issues arise. Thanks!

closed 2024-07-23 by twslankard