ouster-lidar/ouster-sdk #583: Lots of black strip and keep flashing when recording

Dolan-Liu, 2024-03-25:
Whenever I attempt a recording, the screen flickers continuously, and numerous black stripes appear. If anyone has encountered a similar issue or has suggestions for a resolution, could you please share your insights here? Thanks!
[image]

yucedagonurcan, 2024-04-26:
It can be due to bandwidth, loss of packages from network transport, do check out other issues and you can also try the lighter UDP profiles to check if it helps

laurent82, 2024-05-13:
Hello
I'm also facing this problem on some computers (and no problem at all on others with same specs). The only hint I have is indeed it is dropping a lot of packets for some reasons. 

This is my configuration, using Python and SDK 0.10.0 (I haven't tested yet with version 0.11.0)
`       sensor_config.operating_mode = client.OperatingMode.OPERATING_NORMAL
        sensor_config.lidar_mode = client.LidarMode.MODE_1024x10
        sensor_config.udp_port_lidar = 7502
        sensor_config.udp_port_imu = 7503
        sensor_config.signal_multiplier = 1
        sensor_config.azimuth_window = (0, 360000)
        sensor_config.timestamp_mode = client.TimestampMode.TIME_FROM_SYNC_PULSE_IN
        sensor_config.multipurpose_io_mode = client.MultipurposeIOMode.MULTIPURPOSE_INPUT_NMEA_UART
        sensor_config.nmea_baud_rate = client.NMEABaudRate.BAUD_115200
        sensor_config.nmea_leap_seconds = 0
        sensor_config.phase_lock_enable = False
        sensor_config.sync_pulse_out_pulse_width = 10
`
Computers that have the problem are running on Windows 10 or 11. They do have a gigabit ethernet. I tried to play with the different settings of the network adapter (energy-saving mode or related) but I didn't find anything relevant. The problem also happens randomly.

Can you elaborate about the lighter UDP profiles?

yucedagonurcan, 2024-05-13:
@laurent82  you can check out the following: https://static.ouster.dev/sensor-docs/image_route1/image_route3/sensor_data/sensor-data.html#id17

You can set it via launch file(if ROS): https://github.com/ouster-lidar/ouster-ros/blob/8dd5f73b9f16d30a4071b78a2327d3387fb142a8/launch/driver.launch#L8-L13

matt-attack, 2024-11-09:
For Windows this post has some instructions on mitigations: https://community.ouster.com/t/increase-receive-buffer-to-eliminate-packet-drop-problem-on-some-windows-computers/84