ouster-lidar/ouster-sdk #588: Next release?

furushchev, 2024-04-24:
Since there are some important bugfix since the last release, I'd like to request to release a new version of this repository.
The unreleased changes includes the mandatory changes for building the official downstream package succesfully: https://github.com/ouster-lidar/ouster-ros:

https://github.com/ouster-lidar/ouster_example/pull/566

Samahu, 2024-04-24:
@furushchev good news we do have a release due today

furushchev, 2024-04-25:
@Samahu Glad to know! I'm waiting for the release!

furushchev, 2024-04-25:
@Samahu Hi! Is the release being delayed? If so, can you re-open the issue, please?

Samahu, 2024-04-25:
It is almost there: https://github.com/ouster-lidar/ouster_example/pull/589 we had some minor detail to change but it should be out soon (including the issue you brought up).

Samahu, 2024-04-25:
https://github.com/ouster-lidar/ouster_example/pull/590