ouster-lidar/ouster-sdk #609: read IMU data(timestamp, angular velocity, linear acceleration)

moonmoonmoonmoon, 2024-09-02:
Hello. How can I use the Python API to read IMU data, such as linear acceleration, angular velocity, and timestamp information?

Samahu, 2024-09-03:
It isn't currently possible to get imu packets using current OusterSDK Python API, we are adding support for that sometime in next release

Samahu, 2024-09-03:
You could use the steps described here https://static.ouster.dev/sdk-docs/python/examples/udp-packets.html to access ImuPackets but instead of constructing a `source = pcap.Pcap(pcap_path, metadata)` object you would need to use `source = Sensor(source_name, 7502, 7503)` if you want to read the imu data from a live sensor.

Note that the `pcap.Pcap` and `client.Sensor` are part of legacy that will be removed in the future. We will have a wholesome approach to read imu that is consistent with the more `ScanSource` interface https://static.ouster.dev/sdk-docs/python/quickstart.html#using-the-scansource-interface sometime in the near future.

closed 2024-09-03 by Samahu

moonmoonmoonmoon, 2024-09-05:
> You could use the steps described here https://static.ouster.dev/sdk-docs/python/examples/udp-packets.html to access ImuPackets but instead of constructing a `source = pcap.Pcap(pcap_path, [...]

Thank you su much!