ouster-lidar/ouster-sdk #612: Program crashes when trying to create a osf writer

joaocalen, 2024-09-09:
**Bug**
The bug consists when trying to instantiate a Writer object. I get the metadata from the OS1 ouster and create a sensor::sensor_info structure from it. But when I try to create a Writer using this sensor_info, the program crashes.

**Code:**

`    sensor::init_logger("info", "ouster.log");

    std::cerr << "Ouster client example " << ouster::SDK_VERSION << std::endl;
    
    const std::string sensor_hostname = argv[1];
    const std::string data_destination = (argc == 3) ? argv[2] : "";

    std::cerr << "Connecting to \"" << sensor_hostname << "\"...\n";

    auto handle = sensor::init_client(sensor_hostname, data_destination);
    if (!handle) FATAL("Failed to connect");
    std::cerr << "Connection to sensor succeeded" << std::endl;

    std::cerr << "Gathering metadata..." << std::endl;
    std::string metadata = sensor::get_metadata(*handle, 10);

    // // Raw metadata can be parsed into a `sensor_info` struct
    sensor::sensor_info info(metadata);   

    const std::string osf_file = "teste.osf"; 
    
    osf::Writer writer(osf_file, info);
   
    size_t w = info.format.columns_per_frame;
    size_t h = info.format.pixels_per_column;

    ouster::sensor::ColumnWindow column_window = info.format.column_window;

    std::cerr << "  Firmware version:  " << info.fw_rev
              << "\n  Serial number:     " << info.sn
              << "\n  Product line:      " << info.prod_line
              << "\n  Scan dimensions:   " << w << " x " << h
              << "\n  Column window:     [" << column_window.first << ", "
              << column_window.second << "]" << std::endl;

`

**Output**
Ouster client example 0.12.0
Connecting to "192.168.1.200"...
Connection to sensor succeeded
Gathering metadata...
munmap_chunk(): invalid pointer
Aborted (core dumped)
------------------------------

**Sensor info**
    {
        "build_date": "2024-01-11T06:02:47Z",
        "build_rev": "v2.5.3",
        "image_rev": "ousteros-image-prod-aries-v2.5.3+20240111055903",
        "initialization_id": 7109745,
        "prod_line": "OS-1-32-U3",
        "prod_pn": "840-105461-06",
        "prod_sn": "122307000631",
        "status": "RUNNING"
    },
--------------------------------------

**Aditional Output (with valgrind)**
==6132== Command: ./lidar_visualize 192.168.1.200
==6132== 
Ouster client example 0.12.0
Connecting to "192.168.1.200"...
Connection to sensor succeeded
[... 65 lines cut ...]
==6132==    by 0x23A00B: ouster::osf::Writer::~Writer() (in /root/drone-logger/build/bin/lidar_visualize)
==6132==    by 0x12F746: main (in /root/drone-logger/build/bin/lidar_visualize)
==6132== 
==6132== Invalid free() / delete / delete[] / realloc()
==6132==    at 0x484BB6F: operator delete(void*, unsigned long) (in /usr/libexec/valgrind/vgpreload_memcheck-amd64-linux.so)
==6132==    by 0x1342ED: ouster::sensor::sensor_info::~sensor_info() (in /root/drone-logger/build/bin/lidar_visualize)
==6132==    by 0x12F771: main (in /root/drone-logger/build/bin/lidar_visualize)
==6132==  Address 0x1 is not stack'd, malloc'd or (recently) free'd
==6132== 
==6132== 
==6132== HEAP SUMMARY:
==6132==     in use at exit: 60 bytes in 3 blocks
==6132==   total heap usage: 25,381 allocs, 25,382 frees, 1,558,699 bytes allocated
==6132== 
==6132== LEAK SUMMARY:
==6132==    definitely lost: 60 bytes in 3 blocks
==6132==    indirectly lost: 0 bytes in 0 blocks
==6132==      possibly lost: 0 bytes in 0 blocks
==6132==    still reachable: 0 bytes in 0 blocks
==6132==         suppressed: 0 bytes in 0 blocks
==6132== Rerun with --leak-check=full to see details of leaked memory
==6132== 
==6132== Use --track-origins=yes to see where uninitialised values come from
==6132== For lists of detected and suppressed errors, rerun with: -s
==6132== ERROR SUMMARY: 6 errors from 6 contexts (suppressed: 0 from 0)

diego-guridi, 2024-09-09:
Hi @joaocalen 

Can you please share the sensor_info you get? you could serialize it using:

```
sensor_info.to_json_string();
```

joaocalen, 2024-09-10:
Sure. Here is the output:

Gathering metadata...
{
    "beam_intrinsics": 
    {
        "beam_altitude_angles": 
        [
            20.63,
            19.38,
            18.14,
            16.85,
            15.56,
            14.26,
            12.93,
            11.59,
            10.24,
            8.87,
            7.5,
            6.12,
            4.73,
            3.36,
            1.96,
            0.55,
            -0.86,
            -2.25,
            -3.64,
            -5.03,
            -6.42,
            -7.8,
            -9.16,
            -10.53,
            -11.88,
            -13.22,
            -14.52,
            -15.82,
            -17.1,
            -18.36,
            -19.6,
            -20.83
        ],
        "beam_azimuth_angles": 
        [
            4.18,
            4.19,
            4.19,
            4.19,
            4.2,
            4.2,
            4.2,
            4.2,
            4.2,
            4.18,
            4.19,
            4.18,
            4.2,
            4.2,
            4.19,
            4.19,
            4.19,
            4.2,
            4.19,
            4.19,
            4.2,
            4.21,
            4.21,
            4.22,
            4.21,
            4.21,
            4.23,
            4.22,
            4.24,
            4.23,
            4.23,
            4.22
        ],
        "beam_to_lidar_transform": 
        [
            1.0,
            0.0,
            0.0,
            15.806,
            0.0,
            1.0,
            0.0,
            0.0,
            0.0,
            0.0,
            1.0,
            0.0,
            0.0,
            0.0,
            0.0,
            1.0
        ],
        "lidar_origin_to_beam_origin_mm": 15.806
    },
    "calibration_status": 
    {
        "reflectivity": 
        {
            "timestamp": "2023-02-18T14:38:41",
            "valid": true
        }
    },
    "config_params": 
    {
        "azimuth_window": 
        [
            0,
            360000
        ],
        "columns_per_packet": 16,
        "lidar_mode": "1024x10",
        "multipurpose_io_mode": "OFF",
        "nmea_baud_rate": "BAUD_9600",
        "nmea_ignore_valid_char": 0,
        "nmea_in_polarity": "ACTIVE_HIGH",
        "nmea_leap_seconds": 0,
        "operating_mode": "NORMAL",
        "phase_lock_enable": false,
        "phase_lock_offset": 0,
        "signal_multiplier": 1,
        "sync_pulse_in_polarity": "ACTIVE_HIGH",
        "sync_pulse_out_angle": 360,
        "sync_pulse_out_frequency": 1,
        "sync_pulse_out_polarity": "ACTIVE_HIGH",
        "sync_pulse_out_pulse_width": 10,
        "timestamp_mode": "TIME_FROM_INTERNAL_OSC",
        "udp_dest": "192.168.1.1",
        "udp_port_imu": 58009,
        "udp_port_lidar": 60501,
        "udp_profile_imu": "LEGACY",
        "udp_profile_lidar": "RNG19_RFL8_SIG16_NIR16"
    },
    "imu_intrinsics": 
    {
        "imu_to_sensor_transform": 
        [
            1.0,
            0.0,
            0.0,
            6.253,
            0.0,
            1.0,
            0.0,
            -11.775,
            0.0,
            0.0,
            1.0,
            7.645,
            0.0,
            0.0,
            0.0,
            1.0
        ]
    },
    "lidar_data_format": 
    {
        "column_window": 
        [
            0,
            1023
        ],
        "columns_per_frame": 1024,
        "columns_per_packet": 16,
        "fps": 10,
        "pixel_shift_by_row": 
        [
            12,
            12,
            12,
            12,
            12,
            12,
            12,
            12,
            12,
            12,
            12,
            12,
            12,
            12,
            12,
            12,
            12,
            12,
            12,
            12,
            12,
            12,
            12,
            12,
            12,
            12,
            12,
            12,
            12,
            12,
            12,
            12
        ],
        "pixels_per_column": 32,
        "udp_profile_imu": "LEGACY",
        "udp_profile_lidar": "RNG19_RFL8_SIG16_NIR16"
    },
    "lidar_intrinsics": 
    {
        "lidar_to_sensor_transform": 
        [
            -1.0,
            0.0,
            0.0,
            0.0,
            0.0,
            -1.0,
            0.0,
            0.0,
            0.0,
            0.0,
            1.0,
            36.18,
            0.0,
            0.0,
            0.0,
            1.0
        ]
    },
    "ouster-sdk": 
    {
        "client_version": "ouster_client 0.12.0",
        "extrinsic": 
        [
            1.0,
            0.0,
            0.0,
            0.0,
            0.0,
            1.0,
            0.0,
            0.0,
            0.0,
            0.0,
            1.0,
            0.0,
            0.0,
            0.0,
            0.0,
            1.0
        ],
        "output_source": "sensor_info_to_string"
    },
    "sensor_info": 
    {
        "build_date": "2024-01-11T06:02:47Z",
        "build_rev": "v2.5.3",
        "image_rev": "ousteros-image-prod-aries-v2.5.3+20240111055903",
        "initialization_id": 7109745,
        "prod_line": "OS-1-32-U3",
        "prod_pn": "840-105461-06",
        "prod_sn": "122307000631",
        "status": "RUNNING"
    },
    "user_data": ""
}

matt-attack, 2024-09-10:
Just a wild guess, but what C++ version are you building your program with? 

I've seen similar errors when the SDK is built against C++14 and the executable is built with C++17 due to an issue with nostd::optional that will be fixed in the next release.

joaocalen, 2024-09-10:
> Just a wild guess, but what C++ version are you building your program with?
> 
> I've seen similar errors when the SDK is built against C++14 and the executable is built with C++17 due to an issue with nostd::optional that will be fixed in the next release.

That was exactly the problem. Thank you so much. I built both SDK and my project with C++17 and now it works. Thank you.

joaocalen, 2024-09-10:
So, problem solved, closing the issue.

closed 2024-09-10 by joaocalen