ouster-lidar/ouster-sdk #613: split pcap

moonmoonmoonmoon, 2024-09-21:
Why can't some of the pcap files display and loading properly in Ouster Studio after being split using this command（ouster-cli source original.pcap slice 1:100 save split.pcap) ?

twslankard, 2024-09-23:
@moonmoonmoonmoon - thanks for your question.

Which version of `ouster-sdk` and which version of Ouster Studio are you using? Also, can you say whether you obtained the original pcap file using `ouster-sdk` or some other means? Using this information, I will try to replicate the results on my end.

Regards,
Tom

moonmoonmoonmoon, 2024-11-22:
Thank you, I am not sure the means of collecting data(original pcap file). If I collect data in the future and still need to split pcap file and still have the same question, I will contact you. Thank you again.