ouster-lidar/ouster-sdk #617: Obtain every frame IMU's timestamps and IMUPacket's timestamps

moonmoonmoonmoon, 2024-10-16:
Hello, how can I get the IMU timestamps information by using python, such as Accelerometer Read Time and Gyroscope Read Time? And how can I get the IMUPacket's timestamps information?

Thank you!

matthew-lidar, 2024-11-16:
Here's an example script that prints those in Python
```python
from ouster.sdk import client

import sys

config = client.SensorConfig()
config.udp_dest = '@auto'
packets = client.SensorPacketSource([(sys.argv[1], config)])
pf = client.PacketFormat(packets.metadata[0])

for i, pack in enumerate(packets):
    index, packet = pack
    if i >= 100:
        break

    if isinstance(packet, client.ImuPacket):
        ax = pf.imu_la_x(packet.buf)
        ay = pf.imu_la_y(packet.buf)
        az = pf.imu_la_z(packet.buf)
        ats = pf.imu_accel_ts(packet.buf)
        gts = pf.imu_gyro_ts(packet.buf)
        print(ats, gts)

```

moonmoonmoonmoon, 2024-11-22:
Thank you very much!!!

closed 2025-01-27 by Samahu