ouster-lidar/ouster-sdk #619: Socket polling method

outrider-jkoch, 2024-10-18:
Has it ever been considered to use a different polling method besides `select()` such as `poll()` or `epoll()`? I see the implementation changed slightly at version 0.11.0 to start abstracting some stuff out of `poll_client()` and moving it into some implementation methods. Perhaps there aren't any substantial performance gains to be had by changing but from my understanding the Linux community has been slowly moving away from using `select()` in favor of one of the others due to slightly better performance and the ability to handle potentially larger file descriptor values.

matt-attack, 2024-11-09:
Might be worth evaluation, though not sure there's much to gain compared to other optimizations.

Perhaps the main reason we haven't evaluated it so far is it is a platform specific API, but we support both Mac and Windows as well. Also, commonly most of the gains with E-Poll are in cases with large numbers of sockets but there's generally few here. This is especially true with our newer APIs which allow multiple sensors and data streams to share the same ports letting one go with a few as a single socket.

outrider-jkoch, 2024-11-13:
Thanks for the response. I understand the aspects of supporting multiple platforms and also not needed to handle a large numbers of sockets. I haven't studied the code enough to understand potential worst case scenarios and if the reconnection logic could eventually produce file descriptors with values that are out of the currently supported `select()` range.