ouster-lidar/ouster-sdk #620: Expose compile-time SDK version

themightyoarfish, 2024-10-18:
**Is your feature request related to a problem? Please describe.**

The runtime SDK version can be obtained by `ouster::util::version_from_string(ouster::client_version())`, but since sometimes the type of things changes (e.g. the need to use `uint8_t` instead of `uint16_t` to access reflectivity channel), it would be useful to be able to fix this at compile time, based on the SDK version being used.

**Describe the solution you'd like**

Expose `SDK_VERSION` to client code, instead of hiding it inside the library.

**Describe alternatives you've considered**
n/a

**Targeted Platform (please complete the following information only if applicable, otherwise dot N/A):**
any

themightyoarfish, 2024-10-18:
> Maybe I am missing something but the environment variable is already exposed via `ouster::SDK_VERSION` or  `ouster::SDK_VERSION_FULL` you need to include `#include <ouster/impl/build.h>` to access it.

Huh, are the impl headers even installed? My assumption was that impl headers are not supposed to be used by clients.

Samahu, 2024-10-18:
Yes they are installed, and you are free to use them

themightyoarfish, 2024-10-21:
Ok, i was not aware that this was intended.

closed 2024-10-21 by themightyoarfish