ouster-lidar/ouster-sdk #621: Gaps in PointCloud read from Pcap file using the c++ api

hirenpatel1207, 2024-10-22:
**Describe the bug**
When reading Pcap file by using the c++ api. The resulting pointcloud has gap as shown in the figure.

Used the ouster-sdk version (release-0.13.0) commit.

**To Reproduce**
Steps to reproduce the behavior (steps below are just an example):
1. Use ouster studio or pcap example to record Pcap and meta data file.
2. Use the below code to read the Pcap file.
3. View the stored point data using any library like Pcl, Open3d etc.
4. See the screenshots

**Scree
[image]
nshots**

[image]

Targeted Platform (please complete the following information only if applicable, otherwise dot N/A):

Ouster Sensor: OS-1
Ouster Firmware Version? [feature only applicable to specific fw version]
Programming Language C++
Operating System: Windows
Machine Architecture x86

In issue https://github.com/ouster-lidar/ouster-sdk/issues/611#issue-2513937756
@diego-guridi suggest me code to read an write Pcap files.
At first I used the sample code to record Pcap file. 
Then I used the next example which show how to read the Pcap file.

I modified the code to store the read PointCloud data to a csv file as follows:
`#include "ouster/os_pcap.h"
#include "ouster/lidar_scan.h"

int main() {
    const std::string pcap_file = "my_pcap_file.pcap";
    const std::string json_file = "my_pcap_file.json";

    auto metadata = ouster::sensor::metadata_from_json(json_file);
    ouster::sensor::Packet packet(packet_format.lidar_packet_size);
    ouster::sensor_utils::packet_info packet_info;

    auto width = metadata.format.columns_per_frame;
    auto height = metadata.format.pixels_per_column;
    ouster::LidarScan lidar_scan = ouster::LidarScan(width, height, metadata.format.udp_profile_lidar);
    ouster::ScanBatcher scan_batcher(metadata.format.columns_per_frame, packet_format);

    auto pcap_handle = ouster::sensor_utils::replay_initialize(pcap_file);
    
    static scanIndex=1;

     while (ouster::sensor_utils::next_packet_info(*pcap_handle, packet_info)) {
        auto packet_size = ouster::sensor_utils::read_packet(*pcap_handle, packet.buf.data(), packet.buf.size());

        if (packet_size == packet_format.lidar_packet_size) {
            const auto &lidar_packet = static_cast<ouster::sensor::LidarPacket &>(packet);
            if (scan_batcher(lidar_packet, lidar_scan)) {
            
            scanIndex++;
            
            if (scanIndex == 10)
            {
            out.open("scan_" + std::to_string(time++) + ".csv");
            out << std::fixed << std::setprecision(4);
            XYZLut xyz_lut = make_xyz_lut(info, true);
            // storing the 10th lidar scan to a file and stopping
            ouster::LidarScan::Points scanCloud = cartesian(lidar_scan, xyz_lut);
            for (int i = 0; i < scanCloud.rows(); i++)
               {
               out << xyz(0) << ", " << xyz(1) << ", " << xyz(2) << std::endl;
               }
            out.close();
            std::cerr << "  Wrote " << filename << std::endl;
            break;            
            }

            }
        }
     }
}`

But the scan has gaps as seen in the figure. I then opened the Pcap file in Ouster Studio and there, they look fine (2nd screenshot).
Then I thought that the c++ program to write the Pcap might have some problem, so I used OusterStudio to record Pcap file and then use the c++ program to read and save a csv file from it. I go the same problem. 

I can share the recorded Pcap if required.

twslankard, 2024-10-22:
@hirenpatel1207 

Thanks for your bug report. Unfortunately I am unable to compile the example you provided. When convenient, would you be able to provide the complete implementation, plus the pcap and metadata JSON file from your example?

Best Regards,
Tom

Staff Engineer
Ouster Inc.

twslankard, 2024-10-22:
@hirenpatel1207 

I was able to modify your example somewhat to get it working. I was unable to reproduce the issue from pcap data recorded using `ouster-cli` or Ouster Studio 2.0.5.

Bear in mind that the first scan or two may be incomplete (depending on when the client socket starts reading from the sensor during recording.) Also, scans may be incomplete if the sensor's azimuth window is configured to be anything other than [0, 360000].

Here is a gist of the code I used to attempt to reproduce the issue.
https://gist.github.com/twslankard/f88d01bcaba64b4b3d1b6bc5bdfbd4e2

The code prints the scan number followed by the number of packets per scan. Since the default number of columns per packet is 16, one can determine whether any packets were dropped (or simply not recorded depending on when the recording was started, as I mention above.)

For a full azimuth window in 1024x10 mode for example, there should be 1024/16 == 64 packets per scan.

hirenpatel1207, 2024-10-23:
@twslankard  Thanks for the solution.
The sample code you provided worked fine. So I went back to my code and tried to find out the root of the problem.
In my project, I run the PcapReader in a seperate thread. On which I regularly send a Signal to read the next point cloud.
What was happening was the   **ouster::LidarScan lidar_scan** and  **ouster::ScanBatcher scan_batcher** were getting out of scope as the method finishes reading one scan. Then when i call the method again, they get recreated again. I moved those variables to class members so that they are initialized only once in the constructor. And then it worked.
But it does raise the question that does **ScanBatcher** has some internal state?

twslankard, 2024-10-23:
Happy to help!

Regarding `ScanBatcher` - yes, the batcher only accepts a packet at a time so it necessarily tracks whether the packet provided to the `operator()` method belongs to the current `LidarScan` or not. A stateless design is likely possible, but it is not on our roadmap.

Cheers,
Tom

hirenpatel1207, 2024-10-24:
Alright, understood. Closing this issue.

closed 2024-10-24 by hirenpatel1207