ouster-lidar/ouster-sdk #625: Question about using PointViz to draw unstructured point clouds

dwniaoniao, 2024-11-14:
This is my code:
```
from contextlib import closing
from ouster.sdk import open_source
from ouster.sdk import client
from ouster.sdk import viz
import numpy as np

def get_xyz(scan, metadata, destagger = False):
    xyzlut = client.XYZLut(metadata)
    if not destagger:
        xyz = xyzlut(scan)
    else:
        xyz = client.destagger(metadata, xyzlut(scan))
    return xyz

pcap_file = "Ouster-YOLOv5-sample.pcap"
with closing(open_source(pcap_file, index = True)) as source:
    scan = source[10]
    xyz = get_xyz(scan, source.metadata, False)
    print(xyz.shape, xyz.dtype)

    # Creating a point viz instance
    point_viz = viz.PointViz("Example Viz")
    viz.add_default_controls(point_viz)
    
    # ... add objects here
    cloud_xyz = viz.Cloud(xyz.shape[0] * xyz.shape[1])
    cloud_xyz.set_xyz(xyz.reshape(-1, 3))
    point_viz.add(cloud_xyz)
    
    # update internal objects buffers and run visualizer
    point_viz.update()
    point_viz.run()
```
It runs well and the draw point cloud looks OK, then i change the line:
`xyz = get_xyz(scan, source.metadata, False)`
To
`xyz = get_xyz(scan, source.metadata, True)`
Then the draw point cloud looks completely disorganized, but the array
xyz return by `get_xyz(scan, source.metadata, True)` can be normaly draw
by using Open3d. Is that means when using PointViz to drow point cloud
the point cloud array should meet some certian requirements?
The outputs point clouds as bellow, the first one is normal and the second
is not:
[image]
[image]

matthew-lidar, 2024-11-14:
Hi,

The issue is the `set_xyz` function currently expects the data for the points in a strange order, first all X's then all Y's then all Z's. This will be improved in the next release so that it can convert from any ordering as necessary. Until then you just need to re-order the points.

Reshaping with 'F' ordering should do the trick here:

```
cloud_xyz.set_xyz(xyz.reshape((-1, 3), order='F'))
```

dwniaoniao, 2024-11-15:
Thank you @[matthew-lidar](https://github.com/matthew-lidar), problem solved.

closed 2024-11-15 by dwniaoniao