ouster-lidar/ouster-sdk #626: Providing metadata path in open_source args

AbhiKhoyani, 2024-11-14:
**Is your feature request related to a problem? Please describe.**
Can you please add `meta:str=<<metadata.json path>>` argument in `ouster.sdk.opens_source` function so that we do not need to copy-paste metadata.json file with specific sequence name in specific path all the time. As, we're recording only pointcloud data in pcap format using other recording tool and not by ouster-studio which saves metadata.json everytime recording. 

**Describe the solution you'd like**
Providing meta args in open_source as similar to `PcapScanSource` shown in below.
`front_source = PcapScanSource(front_pcap, meta=[front_config], extrinsics=extrinsics["lidar_front"], soft_id_check=True)`

**Describe alternatives you've considered**
I've tried reading pcap files using PcapScanSource class but it doesn't consider externally provided `extrinsic` arguments. 

**Targeted Platform (please complete the following information only if applicable, otherwise dot N/A):**
N/A

Samahu, 2024-11-14:
The open_source already support passing a specific metadata path in case you needed to.

This should work:
```pyrhon
from ouster.sdk import open_source

source_url = ...
metadata1_path = ...
metadata2_path = ...
scan_source = open_source(source_url, meta=[metadata1_path, metadata2_path, ...], ...)
```

It is not an explicit parameter but it will be forwards to the PcapScanSource.

Samahu, 2024-11-14:
BTW, the open_source will try to resolve the metadata path automatically for you. In most cases this would be the best way to use the method, however you can manually set the metadata path as stated above.

Samahu, 2024-11-14:
Regarding the 2nd part of your request you should also be able to supply the extrinscis to the `open_source` method:

```python
def open_source(source_url: Union[str, List[str]], sensor_idx: int = 0, *args,
                extrinsics: Optional[Union[str, np.ndarray, List[np.ndarray]]] = None,
                **kwargs) -> Union[ScanSource, MultiScanSource]:
```

This expects a 4x4 numpy array representing an extrinisc transform (or a list of them in case you are using the multi-sensor pcap file).

The handling of extriniscs param is carried over within the the `open_source` method.

If you really insist on utilizing the **PcapScanSource** directly you can set the extrinsics as follows:
```python
scan_source = PcapScanSource(...)
scan_source.metadata[0].extrinisc = extrinsics["lidar_front"]
scan_source.metadata[1].extrinisc = extrinsics["back_front"]
...
```

closed 2024-11-14 by Samahu